fkie-cad/FACT_core #558: Install automation with ansible, questions

juju4, 2021-03-28:
Thanks a lot for your work. Great project!

I'm building an ansible role to automate setup of FACT. https://github.com/juju4/ansible-fact/
for now, devel branch is using python installer and working fine in Azure + Ubuntu 18.04 or 20.04. (with some github action only issues)
Still work on Fedora/RHEL side and working on a full ansible install in separate branch without python installer.

I have a few questions
- any way to test submission of example router file like https://github.com/praetorian-inc/DVRF/blob/master/Firmware/DVRF_v03.bin through CLI or API? that would also be useful to submit large firmware locally. tried 1GB TV firmware and did not submit...
Not found relevant example in https://github.com/fkie-cad/FACT_core/wiki/Rest-API. Any curl examples?
- any other tests that would be useful to ensure instance is functional?
- any way to avoid the use of docker privileged in extractor? I put the 3 services under systemd unit and add some seccomp/cgroups hardening but for backend with docker privileged, that reduced a lot what is possible and not seen for now what in extraction requires it.
https://github.com/fkie-cad/FACT_core/blob/master/src/unpacker/unpack_base.py#L23
https://github.com/fkie-cad/fact_extractor/blob/master/extract.py#L79
- on backup & restore, need mongodump, /media/data. anything else?
- export pdf function failed on my side. need to revisit for details.

jstucke, 2021-03-29:
Hi and thank you!

> any way to test submission of example router file like https://github.com/praetorian-inc/DVRF/blob/master/Firmware/DVRF_v03.bin through CLI or API? that would also be useful to submit large firmware locally. tried 1GB TV firmware and did not submit...

Submitting very large firmware images is currently an issue because the request may time out. We are addressing this issue in #552 (at least for the frontend). Did you try the REST endpoint (see https://github.com/fkie-cad/FACT_core/wiki/Rest-API)?

> any other tests that would be useful to ensure instance is functional?

Did you try running pytest? The tests should cover most of the functionalities in FACT.

> any way to avoid the use of docker privileged in extractor?

Docker privileged is needed for mounting file systems in the extractor (in order to unpack them). We haven't found a good way to circumvent this yet.

> on backup & restore, need mongodump, /media/data. anything else?

Maybe the configuration files inside `src/config` if you made changes to them. The config file for FACT (main.cfg) is especially important as it contains the password that is also used for DB authentication.

juju4, 2021-06-13:
Sorry for delayed update.
Yes, REST API was what I was looking for. not sure how I missed it or if added recently.
pytest: again, right what I need. Thanks!
Added both to my verify playbook

on backup/restore, I was more thinking on data/reports because if setup is automated which it is with ansible, config does not really matter.
If many analysis/processing done, we probably want to be able to keep results and restore without reprocessing everything. was more thinking on this side.

dorpvom, 2021-06-14:
Regarding backup, there is an old issue similar to what you intend I think #446.
We haven't addressed this since. I guess it should not be overly complicated writing a naive script to backup data files and database entries recursively. If you are interested in writing something yourself, we could provide some pointers as to where to start and possible pitfalls.

jstucke, 2024-12-04:
Concerning backup: #1278 added a script for importing and exporting files and analysis data.

Since there has been no activity in this issue for years I will close it. If you have more questions, feel free to reopen it.

closed 2024-12-04 by jstucke