fkie-cad/FACT_core #559: Exception: No available plug-ins found. FACT backend might be down! - intermittent

juju4, 2021-03-28:
As part of https://github.com/juju4/ansible-fact/

This testing (devel) works fine on 18.04 and 20.04
https://github.com/juju4/ansible-fact/actions/runs/695544247

and fails in master branch for 20.04. ok 18.04.
https://github.com/juju4/ansible-fact/runs/2213456944?check_suite_focus=true

Error from /var/log/fact/fact_main.log
```
Error:        +[2021-03-28 18:39:52][cwe_checker][ERROR]: Could not get module versions from Bap plugin: 1 (error: Found argument '/bin/true' which wasn't expected, or isn't valid in this context
              +
              +USAGE:
              +    cwe_checker [FLAGS] [OPTIONS] <binary>
              +
              +For more information try --help
              +). I tried the following command: docker run --rm fkiecad/cwe_checker:latest bap /bin/true --pass=cwe-checker --cwe-checker-module-versions
Error:        +[2021-03-28 18:39:54][app][ERROR]: Exception on /system_health [GET]
              +Traceback (most recent call last):/var/log/fact/fact_main.log
              +  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 2447, in wsgi_app
              +    response = self.full_dispatch_request()
              +  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1952, in full_dispatch_request
              +    rv = self.handle_user_exception(e)
              +  File "/usr/local/lib/python3.8/dist-packages/flask_restful/__init__.py", line 272, in error_router
              +    return original_handler(e)
              +  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1821, in handle_user_exception
              +    reraise(exc_type, exc_value, tb)
              +  File "/usr/local/lib/python3.8/dist-packages/flask/_compat.py", line 39, in reraise
              +    raise value
              +  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1950, in full_dispatch_request
              +    rv = self.dispatch_request()
              +  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1936, in dispatch_request
              +    return self.view_functions[rule.endpoint](**req.view_args)
              +  File "./web_interface/security/decorator.py", line 11, in decorated_view
              +    return fn(*args, **kwargs)
              +  File "./web_interface/components/statistic_routes.py", line 42, in _show_system_health
              +    plugin_dict = sc.get_available_analysis_plugins()
              +  File "./intercom/front_end_binding.py", line 36, in get_available_analysis_plugins
              +    raise Exception("No available plug-ins found. FACT backend might be down!")
              +Exception: No available plug-ins found. FACT backend might be down!
```

I tested `docker run --rm fkiecad/cwe_checker:latest bap /bin/true --pass=cwe-checker --cwe-checker-module-versions` in separate live instance and it gives same output without affecting functionality.
Likely different problem.

As problem is intermittent and sometimes appears in 18.04 and not 20.04, I'm uncertain.
I tried to add some extra delay as maybe needed for some services to fully start.

Any idea to debug?

Thanks

jstucke, 2021-03-29:
Hi, it seems like the frontend cannot connect to the backend (usually this is done internally using the "intercom") or that the backend is not running. 

What is your setup? Are you running a distributed setup (i.e. backend, frontend and db running on different systems) or is everything running on the same machine? What script did you use to start FACT?

rhelmke, 2021-12-02:
Hey,

due to the inactivity here I'm going to close this issue. Feel free to re-open in case you are still experiencing the described problem.

closed 2021-12-02 by rhelmke