fkie-cad/FACT_core #561: Problems during install on Ubuntu 20.04 on wsl2

shumtravi, 2021-04-04:
When i try to install FACT_core in single machine setup on Ubuntu 20.04 on wsl2 in Windows10 x64 v10.0.19042 i get these errors:

> [2021-04-04 14:36:26][install][INFO]: FACT Installer 1.2
> [2021-04-04 14:36:26][common][INFO]: Updating system
> [2021-04-04 14:36:26][install][INFO]: Installing apt-transport-https
> [sudo] password for shum: 
> [2021-04-04 14:36:40][install][INFO]: Installing python3 python3-dev build-essential automake autoconf libtool git unzip
> [2021-04-04 14:36:40][install][INFO]: Installing testresources
> [2021-04-04 14:36:42][install][INFO]: Removing python3-pip python3-setuptools python3-wheel
> [2021-04-04 14:36:42][common][INFO]: Installing python3 pip
> [2021-04-04 14:36:47][install][INFO]: Installing setuptools
> [2021-04-04 14:36:49][install][INFO]: Installing libmagic-dev
> [2021-04-04 14:36:50][install][INFO]: Installing libfuzzy-dev
> [2021-04-04 14:36:50][install][INFO]: Installing python3-tlsh
> [2021-04-04 14:36:51][install][INFO]: Installing ssdeep
> [2021-04-04 14:36:52][install][INFO]: Installing git+https://github.com/fkie-cad/fact_helper_file.git
> [2021-04-04 14:36:54][install][INFO]: Installing psutil pytest pytest-cov pylint python-magic xmltodict yara-python==3.7.0 appdirs flaky
> [2021-04-04 14:37:07][install][INFO]: Installing lief==0.10.1
> [2021-04-04 14:37:09][install][INFO]: Installing requests
> [2021-04-04 14:37:10][install][INFO]: Installing pymongo pyyaml
> [2021-04-04 14:37:13][install][WARNING]: Could not update python package pyyaml. Was not installed using pip originally
> [2021-04-04 14:37:13][common][WARNING]: variety spec not overwritten
> [2021-04-04 14:37:13][install][INFO]: Installing hurry.filesize
> [2021-04-04 14:37:14][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_files.git
> [2021-04-04 14:37:17][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_mongo.git
> [2021-04-04 14:37:19][install][INFO]: Installing git+https://github.com/mass-project/common_helper_encoder.git
> [2021-04-04 14:37:21][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_filter.git
> [2021-04-04 14:37:23][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_process.git
> [2021-04-04 14:37:25][install][INFO]: Installing werkzeug==0.16.1
> [2021-04-04 14:37:27][install][INFO]: Installing flask flask_restful flask_security flask_sqlalchemy flask-paginate Flask-API uwsgi bcrypt python-dateutil si-prefix "sqlalchemy>=1.2,<1.4" email-validator matplotlib
> [2021-04-04 14:37:43][frontend][INFO]: Install static jstree content
> [2021-04-04 14:37:45][frontend][INFO]: Install static angularJS content
> [2021-04-04 14:37:45][frontend][INFO]: Install static charts.js content
> [2021-04-04 14:37:45][frontend][INFO]: Install static vis-network content
> [2021-04-04 14:37:45][frontend][INFO]: Installing highlight js
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/css/bootstrap.min.css content
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/bootstrap-datepicker/1.8.0/css/bootstrap-datepicker.standalone.css content
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/jquery/1.12.1/jquery.min.js content
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/popper.js/1.14.7/umd/popper.min.js content
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/js/bootstrap.min.js content
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/bootstrap-datepicker/1.8.0/js/bootstrap-datepicker.js content
> [2021-04-04 14:37:49][frontend][INFO]: Install static https://raw.githubusercontent.com/moment/moment/develop/moment.js content
> [2021-04-04 14:37:49][frontend][INFO]: Creating directory for authentication
> [2021-04-04 14:37:49][frontend][INFO]: Initializing docker container for radare
> [2021-04-04 14:37:56][frontend][INFO]: Pulling pdf report container
> [2021-04-04 14:37:59][db][INFO]: Setting up mongo database
> [2021-04-04 14:37:59][install][INFO]: Installing mongodb
> [2021-04-04 14:38:00][db][INFO]: Initialize database
> [2021-04-04 14:38:02][install][INFO]: Installing python-dev python-setuptools
> [2021-04-04 14:38:03][install][INFO]: Installing libjpeg-dev
> [2021-04-04 14:38:03][install][INFO]: Installing libssl-dev python3-tk
> [2021-04-04 14:38:04][install][INFO]: Installing pluginbase Pillow cryptography pyopenssl matplotlib docker networkx
> [2021-04-04 14:38:17][backend][INFO]: Installing yara
> [2021-04-04 14:38:17][install][INFO]: Installing bison flex
> [2021-04-04 14:38:17][backend][INFO]: skipping yara installation (already installed)
> [2021-04-04 14:38:17][backend][INFO]: Building fact extraction container
> [2021-04-04 14:38:21][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_yara.git
> [2021-04-04 14:38:24][install][INFO]: Installing git+https://github.com/mass-project/common_analysis_base.git
> [2021-04-04 14:38:26][backend][INFO]: Installing plugins
> [2021-04-04 14:38:26][backend][INFO]: Running ../plugins/analysis/crypto_hints/install.sh
> [2021-04-04 14:38:26][backend][INFO]: Running ../plugins/analysis/oms/install.sh
> [2021-04-04 14:38:41][backend][INFO]: Running ../plugins/analysis/cve_lookup/install.sh
> [2021-04-04 14:39:21][backend][INFO]: Running ../plugins/analysis/qemu_exec/install.sh
> Traceback (most recent call last):
>   File "./install.py", line 177, in <module>
>     install()
>   File "./install.py", line 165, in install
>     backend(distribution)
>   File "/home/shum/FACT_core/src/install/backend.py", line 45, in main
>     _install_plugins(distribution)
>   File "/home/shum/FACT_core/src/install/backend.py", line 109, in _install_plugins
>     raise InstallationError('Error in installation of {} plugin\n{}'.format(Path(install_script).parent.name, shell_output))
> helperFunctions.install.InstallationError: Error in installation of qemu_exec plugin
> Error: docker daemon not running! Could not build docker image

P.S. Docker for windows with wsl2 support installed, and FACT installer successfully installed 3 containers radare_server, fkiecad/fact_pdf_report and fkiecad/fact_extractor

dorpvom, 2021-04-07:
Hi,
we'll look into this. Last time we checked WSL there was no support for loop devices, making some of the analyses and the extraction of multiple container types impossible. I'm not sure if these limitations have been mitigated in the meantime.
We'll update this issue once we had a closer look into this.

jstucke, 2021-04-12:
> helperFunctions.install.InstallationError: Error in installation of qemu_exec plugin
> Error: docker daemon not running! Could not build docker image

The build step that failed seems to be the `docker build` from the "qemu exec" plugin installation. Does `docker build` work in other cases for you?

maringuu, 2021-04-16:
> The build step that failed seems to be the `docker build` from the "qemu exec" plugin installation. Does `docker build` work in other cases for you?

I had a similar issue when trying do build a docker image of FACT_core. For me the problem was the way the qemu_exec plugin tries to determine whether docker is running or not.
I haven't used WSL but it looks like the same bug.

@shumtravi Can you try the following patch?

```patch
diff --git a/src/plugins/analysis/qemu_exec/code/qemu_exec.py b/src/plugins/analysis/qemu_exec/code/qemu_exec.py
index bc1b9e39..67200996 100644
--- a/src/plugins/analysis/qemu_exec/code/qemu_exec.py
+++ b/src/plugins/analysis/qemu_exec/code/qemu_exec.py
@@ -353,5 +353,5 @@ def merge_identical_results(results: Dict[str, Dict[str, str]]):
 
 
 def docker_is_running() -> bool:
-    _, return_code = execute_shell_command_get_return_code('pgrep dockerd')
+    _, return_code = execute_shell_command_get_return_code('docker info')
     return return_code == 0
diff --git a/src/plugins/analysis/qemu_exec/install.sh b/src/plugins/analysis/qemu_exec/install.sh
index 5bae9b21..a911080e 100755
--- a/src/plugins/analysis/qemu_exec/install.sh
+++ b/src/plugins/analysis/qemu_exec/install.sh
@@ -3,7 +3,7 @@
 cd "$( dirname "${BASH_SOURCE[0]}" )" || exit 1
 
 # build docker container
-if [[ $(pgrep dockerd) ]]; then
+if docker info > /dev/null 2&>1 ; then
     (cd docker && docker build --build-arg=http{,s}_proxy --build-arg=HTTP{,S}_PROXY -t fact/qemu:latest .) || exit 1
 else
     echo "Error: docker daemon not running! Could not build docker image"
```

jstucke, 2021-04-19:
If this fixes the problems, we should probably also fix it in the master branch. Would you like to create a pull request?

rhelmke, 2021-12-02:
Hey,

given the FACT (heh), that we recently merged PR #698 by @timkoers and PR #567 by @maringuu has long been merged too, I consider this issue to be resolved. You can certainly reopen it if you disagree.

closed 2021-12-02 by rhelmke