fkie-cad/FACT_core #576: Install error due to recently updated python packages (flask, itsdangerous)

stl314159, 2021-05-13:
I attempted to install a new FACT_core system today from both master and the latest release (v3.1.3).  It appears as though the flask and itsdangerous packages updated on 5/11 and are causing an issue with a fresh install.  Both issues were resolved by rolling back to a previous pip package version.

```
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 28, in <module>
    from web_interface.frontend_main import WebFrontEnd
  File "./web_interface/frontend_main.py", line 14, in <module>
    from web_interface.components.plugin_routes import PluginRoutes
  File "./web_interface/components/plugin_routes.py", line 6, in <module>
    from flask_restful import Resource
  File "/usr/local/lib/python3.8/dist-packages/flask_restful/__init__.py", line 14, in <module>
    from flask.helpers import _endpoint_from_view_func
ImportError: cannot import name '_endpoint_from_view_func' from 'flask.helpers' (/usr/local/lib/python3.8/dist-packages/flask/helpers.py)
unable to load app 0 (mountpoint='') (callable not found or import error)
```
I resolved this issue by `sudo pip3 uninstall flask` and then `sudo pip3 install flask==1.1.2`

```
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 74, in <module>
    web_interface = WebFrontEnd(config=config)
  File "./web_interface/frontend_main.py", line 26, in __init__
    self._setup_app()
  File "./web_interface/frontend_main.py", line 34, in _setup_app
    user_db, user_interface = add_flask_security_to_app(self.app, self.config)
  File "./web_interface/security/authentication.py", line 18, in add_flask_security_to_app
    security = Security(app, user_interface)
  File "/usr/local/lib/python3.8/dist-packages/flask_security/core.py", line 469, in __init__
    self._state = self.init_app(app, datastore, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/flask_security/core.py", line 495, in init_app
    state = _get_state(app, self.datastore,
  File "/usr/local/lib/python3.8/dist-packages/flask_security/core.py", line 334, in _get_state
    remember_token_serializer=_get_serializer(app, 'remember'),
  File "/usr/local/lib/python3.8/dist-packages/flask_security/core.py", line 319, in _get_serializer
    return URLSafeTimedSerializer(secret_key=secret_key, salt=salt)
  File "/usr/local/lib/python3.8/dist-packages/itsdangerous/serializer.py", line 104, in __init__
    self.secret_keys: _t.List[bytes] = _make_keys_list(secret_key)
  File "/usr/local/lib/python3.8/dist-packages/itsdangerous/signer.py", line 64, in _make_keys_list
    return [want_bytes(s) for s in secret_key]
TypeError: 'NoneType' object is not iterable
unable to load app 0 (mountpoint='') (callable not found or import error)
```
I resolved this issue by `sudo pip3 uninstall itsdangerous` and then `sudo pip3 install itsdangerous==1.1.0`

jstucke, 2021-05-17:
Thank you for reporting this and providing a fix. #577 should provide a temporary fix

closed 2021-05-17 by jstucke

chenjianquan7, 2022-05-11:
its an big keng

PR #577 (2021-05-17): 576 flask version bugfix
resolves #576 
- pin `flask` and `itsdangerous` because of incompatibilities with `flask_restful` and `flask_security`