fkie-cad/FACT_core #578: I click upload and it shows Internal Server Error. How to solve it

13724460607, 2021-05-21:
[image]

jstucke, 2021-05-21:
There should be an error message (and probably a stack trace) in the output on the command line. Could you post that here? Also, what version are you using? The release version or the latest master branch?

13724460607, 2021-05-21:
The master branch is used. Where can I download the distribution

13724460607, 2021-05-21:
[image]

13724460607, 2021-05-21:
[2021-05-21 06:22:51][db_interface_statistic][ERROR]: Could not store statistic database: DuplicateKeyError - E11000 duplicate key error collection: fact_stats.statistic index: _id_ dup key: { : "database" }, full error: {'index': 0, 'code': 11000, 'errmsg': 'E11000 duplicate key error collection: fact_stats.statistic index: _id_ dup key: { : "database" }'}

dorpvom, 2021-05-21:
Hi. It looks like FACT is already running in the background (hence, can't bind socket to port 5000). For a quick fix you can try killing all uwsgi and python3 processes (i.e. `sudo killall -9 uwsgi && sudo killall -9 python3`) and then start FACT anew. Note that for debugging I'd recommend starting the components independent, so the logs can be viewed separately. Use the `./start_fact_{db,backend,frontend}` scripts for that.

dorpvom, 2021-05-21:
BTW Using the master branch is recommended. There will be a release as soon as today but still FACT is mainly tested on the master branch, and not on the releases.

13724460607, 2021-05-21:
OK, thanks. I'll try

13724460607, 2021-05-21:
[image]
Hello, may I ask where I can download this uploaded plug-in

dorpvom, 2021-05-21:
Sorry for the incomprehensible errors. So this states - if you have years of experience looking at FACT errors - that the upload page ( */upload* ) could not be rendered based on the FACT that the backend has never been access, i.e. did not start correctly (there is a longer explanation as to why this needs to have happened).
Have you indeed started the components individually? If so can you also post the backend log? If not please post the complete log of the `start_all_*` script.

dorpvom, 2021-05-21:
Most likely there was a silent error in your FACT installation process that leads to the backend not properly starting. This is a known issue that comes from the many, many dependencies that FACT has (like cve feeds being offline, or, what happens like every two weeks or so, sourceforge being down).

13724460607, 2021-05-21:
I'm wondering where the file for./start_face_backend went;The FACT_core page has only two other files.

dorpvom, 2021-05-21:
Do you have a `start_fact_db` and `start_fact_frontend` and no `start_fact_backend`? This would add to my suspicion that there was an error. In that case please re-run the backend installation `src/install.py -B` (the `-B` flag states that you only want to update the backend and not process the db and frontend dependencies).

closed 2021-05-24 by 13724460607