fkie-cad/FACT_core #579: Repository(s) planned?

SonicFM, 2021-05-21:
Hi! :)

Is there any plan, to create and maintain one or more repository(s) and creating OS-based install files? Like .deb-files and so on.
This would make the install an maintenance of this very great tool much more user-friendly and easly updateable.

If there already is such repo, sry, could you point me towards it?

Appreciating your work!

jstucke, 2021-05-25:
There are currently no such plans, sorry! It would probably be hard to create a package because of all the dependencies (which don't all have packaged versions themselves).

There is a [Vagrant file](https://github.com/botlabsDev/FACTbox), though, for automatically setting up a VM. What is more, we would like to create a FACT docker container in the long run. However, it could take a while before that happens (if it happens).

SonicFM, 2021-05-25:
Thanks for sharing this info. I understand the problems. But it would be great to have a way to make the maintenance of FACT installations easier. I am currently working on a bash-script for Ubuntu 20.04 to make the update process a bit easier, i could share it with you, once i have tested it.

I am really looking forward to see FACT inside a docker container! Hope it happens ;)

SonicFM, 2022-07-24:
https://github.com/fkie-cad/FACT_docker/tree/fact4.0 sounds promising.
Hopefully FACT will be on docker hub at sometime. :)

closed 2022-07-24 by SonicFM