fkie-cad/FACT_core #580: Support for encrypted firmware images / binarys?

SonicFM, 2021-05-21:
Hi! :)

I know it isn't easy to decrypt an encrpyted firmware image / binary, since that is one of the reasons for encrypting and protecting it from beeing manipulated. But i'm wondering if there is any option to implement an automatic or semi-automatic way to help decrypting and afterwards analyzing it.

I was thinking of something like adding an unencrypted version of an image (if available) and then using this to decrypt the newer encrypted one. Like its explained here for example: https://www.zerodayinitiative.com/blog/2020/2/6/mindshare-dealing-with-encrypted-router-firmware

I especialy wondering about the firmware of my Switch (https://www.tp-link.com/en/support/download/t1600g-28ts/#Firmware) and the HP ilo firmwares.

Much thanks in advance!

jstucke, 2021-05-25:
Hi, sadly, I don't think that it is feasible to automate the process, because it usually differs from vendor to vendor (or even from devices to device).  We have an unpacking plugin for encrypted firmware from D-Link (like the firmware in the article you linked) but nothing for TP-Link devices (as for as I know)

You could try to reverse-engineer the process for your device if you have the decrypted firmware (it seems all the ones on the download page are encrypted/compressed). Reversing the process could still be non-trivial even with the decrypted image, though.

SonicFM, 2021-05-25:
Thanks for the reply! Sadly that was the answer i was suspecting, since i know the difficulties. I am not a programmer, so my possibilities in reverse engineering are a "bit" limited. However i will poke around and see what i can find. Thanks again!

rhelmke, 2021-12-02:
Closing as for now.

closed 2021-12-02 by rhelmke