fkie-cad/FACT_core #581: I uploaded the firmware successfully. When I checked the report, it showed an error

13724460607, 2021-05-24:
Please upload the ZIP firmware package. Error is displayed when viewing it.File not found in database: a89650181e54d512cf2484d35c35bb89cd7883707f1205b8facde5cebfa210f8_42811890

1.How can I solve the problem I wrote above
2.Is there any format restriction for uploading firmware package？

13724460607, 2021-05-24:
[image]

jstucke, 2021-05-26:
The way the unpacking process works like this: The unpacking happens in a docker container and the file that is unpacked is mounted into the docker container and the unpacking plugins run inside the container. It seems like the problem in the screenshot is related to the error

> File not found in database: a89650181e54d512cf2484d35c35bb89cd7883707f1205b8facde5cebfa210f8_42811890

It could be a problem with accessing the folder where the files are stored for FACT. The default path is `/media/data/fact_fw_data`. Could you check
- if the directory exists and the user running FACT has access rights
- if the file is present (it shoud have the path `/media/data/fact_fw_data/a8/a89650181e54d512cf2484d35c35bb89cd7883707f1205b8facde5cebfa210f8_42811890`)
- whether other files are present if the file is missing

13724460607, 2021-05-26:
Hello, how can I see the unpacking plugins;
I can see this file.And its path is/media/data/fact_fw_data/a8/a89650181e54d512cf2484d35c35bb89cd7883707f1205b8facde5cebfa210f8_42811890；
How to check this problem with Docker container unpacking

thank

jstucke, 2021-05-27:
Could you try running the extractor container individually? It needs a certain input directory structure, though.

```sh
mkdir -p extractor_test/input extractor_test/reports extractor_test/files
cp /media/data/fact_fw_data/a8/a89650181e54d512cf2484d35c35bb89cd7883707f1205b8facde5cebfa210f8_42811890 extractor_test/input/
docker run --privileged -v /dev:/dev -v $(pwd)/extractor_test:/tmp/extractor --rm fkiecad/fact_extractor
```

After this the extracted output files should normally be in the directory `extractor_test/files`

closed 2021-06-02 by 13724460607