fkie-cad/FACT_core #583: Pinned Version of lief Fails on Kali Rolling

decidedlygray, 2021-05-26:
## Brief Description of Error

I'm working within a relatively new and up-to-date Kali VM. Attempting to install, using the TL;DR instructions in the INSTALL.md I ran into an error running `FACT_core/src/install.py`. I was unable to install the pinned version of the `lief` dependency.

## Workaround
I noticed that the version of `lief` is pinned:
```
pip3_install_packages('lief==0.10.1')  # FIXME: unpin version when install bug is fixed
```
Link to code: https://github.com/fkie-cad/FACT_core/blame/master/src/install/common.py#L72

So I removed the version pin:
```
pip3_install_packages('lief')
```

And everything worked fine. @jstucke I was wondering if you have any background on the FIXME comment? It looks like this came from one of your commits. Am I ok to remove the version pin if the install works, or will I run into issues downstream?

## Ask

Are you able to recreate what I'm seeing? If not I'll close the issue. If yes, and the install works on other compatible OS, I've create a PR here: https://github.com/fkie-cad/FACT_core/pull/582

## Additional Technical Details

Full terminal output:
```
┌──(kali㉿kali)-[/opt/FACT_core]
└─$ ./src/install.py
[2021-05-25 20:48:56][install][INFO]: FACT Installer 1.2
[2021-05-25 20:48:56][common][INFO]: Updating system
[2021-05-25 20:48:56][install][INFO]: Installing apt-transport-https
[2021-05-25 20:49:00][install][INFO]: Installing python3 python3-dev build-essential automake autoconf libtool git unzip
[2021-05-25 20:49:01][install][INFO]: Installing testresources
[2021-05-25 20:49:03][install][INFO]: Removing python3-pip python3-setuptools python3-wheel
[2021-05-25 20:49:03][common][INFO]: Installing python3 pip
[2021-05-25 20:49:09][install][INFO]: Installing setuptools
[2021-05-25 20:49:11][install][INFO]: Installing libmagic-dev
[2021-05-25 20:49:12][install][INFO]: Installing libfuzzy-dev
[2021-05-25 20:49:13][install][INFO]: Installing python3-tlsh
[2021-05-25 20:49:13][install][INFO]: Installing ssdeep
[2021-05-25 20:49:15][install][INFO]: Installing git+https://github.com/fkie-cad/fact_helper_file.git
[2021-05-25 20:49:17][install][INFO]: Installing psutil pytest pytest-cov pylint python-magic xmltodict yara-python==3.7.0 appdirs flaky
[2021-05-25 20:49:33][install][INFO]: Installing lief==0.10.1
Traceback (most recent call last):
  File "/opt/FACT_core/./src/install.py", line 177, in <module>
    install()
  File "/opt/FACT_core/./src/install.py", line 158, in install
    common(distribution)
  File "/opt/FACT_core/src/install/common.py", line 72, in main
    pip3_install_packages('lief==0.10.1')  # FIXME: unpin version when install bug is fixed
  File "/opt/FACT_core/src/helperFunctions/install.py", line 144, in pip3_install_packages
    raise installation_error
  File "/opt/FACT_core/src/helperFunctions/install.py", line 139, in pip3_install_packages
    _run_shell_command_raise_on_return_code('sudo -EH pip3 install --upgrade {}'.format(package), 'Error in installation of python package {}'.format(package), True)
  File "/opt/FACT_core/src/helperFunctions/install.py", line 72, in _run_shell_command_raise_on_return_code
    raise InstallationError(error)
helperFunctions.install.InstallationError: Error in installation of python package lief==0.10.1
Collecting lief==0.10.1
  Using cached lief-0.10.1.tar.gz (12.7 MB)
Building wheels for collected packages: lief
  Building wheel for lief (setup.py): started
  Building wheel for lief (setup.py): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/setup.py'"'"'; __file__='"'"'/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-eyijn0wz
       cwd: /tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/
  Complete output (125 lines):
  0.10.1
  running bdist_wheel
  running build
  running build_ext
[... 242 lines cut ...]
      File "/usr/lib/python3.9/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.9/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.9/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/setup.py", line 78, in run
        self.build_extension(ext)
      File "/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/setup.py", line 202, in build_extension
        subprocess.check_call(configure_cmd, cwd=self.build_temp, env=env)
      File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/build', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DLIEF_PYTHON_API=on', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/setup.py'"'"'; __file__='"'"'/tmp/pip-install-dk2qhgmi/lief_a70adaf234534962a705f25dd6e443d3/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zt_ku5e1/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/lief Check the logs for full command output.

                                                                                                                                                  
┌──(kali㉿kali)-[/opt/FACT_core]
```

Thanks for the great project!

jstucke, 2021-05-27:
> @jstucke I was wondering if you have any background on the FIXME comment?

There was a problem with lief version 0.11.0 when it came out in January. This should hopefully be fixed by now. We will test if on our CI (but that could take a couple of days, because we are currently having some issues with it, incidentally).

jstucke, 2021-05-31:
This should be resolved in #582 so I will close this issue. If something still fails, please feel free to reopen it.

closed 2021-05-31 by jstucke