fkie-cad/FACT_core #585: cwe_checker, cve_checker  rest api no data

logerrors, 2021-06-01:
hi, the webpage has data, but the restful api /rest/firm/<uid> no data.
latest version, 3.2 and ubuntu20.04
[image]
[image]

jstucke, 2021-06-01:
Hi, you need to set the URL parameter "summary" to `true` to get the summary (as documented [here](https://github.com/fkie-cad/FACT_core/wiki/Rest-API#get))

e.g. `curl 'http://localhost:5000/rest/firmware/{{uid}}?summary=true' -X GET`

The summary is not always included because it is dynamically generated from the analysis results of the individual included files (which can be costly and take a while for large firmware).

logerrors, 2021-06-01:
thank you very much , i found it and solve this problem.

closed 2021-06-07 by jstucke