fkie-cad/FACT_core #586: What does the random string mean here

logerrors, 2021-06-02:
[image]
for example, the cwe_checker->summary->CWE119.
the webpage has valid data, but there is not valid data in rest api
[image]

jstucke, 2021-06-02:
That string is the UID ("{MD5}_{filesize}") of a file contained in the firmware with a analysis result of the cwe_checker analysis plugin. The files are unpacked recursively und analyzed individually in FACT. The thing you are looking at in the lower picture is only the summary that is displayed for the firmware as a whole. If you want to get the actual analysis results, you need to retrieve the analysis results of the individual files (i.e. the ones contained in the summary results -- those are the results shown when you click on the links in the summary), e.g. 

```
curl 'http://localhost:5000/rest/file_object/e1e0d7af8df302027ffbee576586a9f41d03a495affadeb8da48eb27453aeef_11692' -X GET
```

rhelmke, 2022-01-24:
Hey, as there haven't been any follow-up questions in the past 6 months, I'm going to close this issue.

Cheers

closed 2022-01-24 by rhelmke