fkie-cad/FACT_core #587: ./install.py安装时一直在这一步停留Pulling pdf report container

13724460607, 2021-06-02:
[2021-06-02 12:40:46][install][INFO]: FACT Installer 1.2
[2021-06-02 12:40:46][common][INFO]: Updating system
[2021-06-02 12:40:46][install][INFO]: Installing apt-transport-https
[2021-06-02 12:40:50][install][INFO]: Installing python3 python3-dev build-essential automake autoconf libtool git unzip
[2021-06-02 12:40:50][install][INFO]: Installing testresources
[2021-06-02 12:40:51][install][INFO]: Removing python3-pip python3-setuptools python3-wheel
[2021-06-02 12:40:51][common][INFO]: Installing python3 pip
[2021-06-02 12:40:58][install][INFO]: Installing setuptools
[2021-06-02 12:40:58][install][INFO]: Installing libmagic-dev
[2021-06-02 12:40:59][install][INFO]: Installing libfuzzy-dev
[2021-06-02 12:40:59][install][INFO]: Installing python3-tlsh
[2021-06-02 12:40:59][install][INFO]: Installing ssdeep
[2021-06-02 12:41:00][install][INFO]: Installing git+https://github.com/fkie-cad/fact_helper_file.git
[2021-06-02 12:41:03][install][INFO]: Installing psutil pytest pytest-cov pylint python-magic xmltodict yara-python==3.7.0 appdirs flaky
[2021-06-02 12:41:08][install][INFO]: Installing lief
[2021-06-02 12:41:09][install][INFO]: Installing requests
[2021-06-02 12:41:10][install][INFO]: Installing pymongo pyyaml
[2021-06-02 12:41:11][install][WARNING]: Could not update python package pyyaml. Was not installed using pip originally
[2021-06-02 12:41:11][common][WARNING]: variety spec not overwritten
[2021-06-02 12:41:11][install][INFO]: Installing hurry.filesize
[2021-06-02 12:41:12][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_files.git
[2021-06-02 12:41:14][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_mongo.git
[2021-06-02 12:41:16][install][INFO]: Installing git+https://github.com/mass-project/common_helper_encoder.git
[2021-06-02 12:41:17][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_filter.git
[2021-06-02 12:41:19][install][INFO]: Installing git+https://github.com/fkie-cad/common_helper_process.git
[2021-06-02 12:41:21][install][INFO]: Installing werkzeug==0.16.1
[2021-06-02 12:41:22][install][INFO]: Installing flask==1.1.2 flask_restful flask_security flask_sqlalchemy flask-paginate Flask-API itsdangerous==1.1.0 uwsgi bcrypt python-dateutil si-prefix "sqlalchemy>=1.2,<1.4" email-validator matplotlib
[2021-06-02 12:41:31][frontend][INFO]: Install static jstree content
[2021-06-02 12:41:34][frontend][INFO]: Install static angularJS content
[2021-06-02 12:41:34][frontend][INFO]: Install static charts.js content
[2021-06-02 12:41:34][frontend][INFO]: Install static vis-network content
[2021-06-02 12:41:34][frontend][INFO]: Installing highlight js
[2021-06-02 12:41:37][frontend][INFO]: Install static https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/css/bootstrap.min.css content
[2021-06-02 12:41:37][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/bootstrap-datepicker/1.8.0/css/bootstrap-datepicker.standalone.css content
[2021-06-02 12:41:37][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/jquery/1.12.1/jquery.min.js content
[2021-06-02 12:41:37][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/popper.js/1.14.7/umd/popper.min.js content
[2021-06-02 12:41:37][frontend][INFO]: Install static https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/js/bootstrap.min.js content
[2021-06-02 12:41:37][frontend][INFO]: Install static https://cdnjs.cloudflare.com/ajax/libs/bootstrap-datepicker/1.8.0/js/bootstrap-datepicker.js content
[2021-06-02 12:41:37][frontend][INFO]: Install static https://raw.githubusercontent.com/moment/moment/develop/moment.js content
[2021-06-02 12:41:37][frontend][INFO]: Creating directory for authentication
[2021-06-02 12:41:37][frontend][INFO]: Initializing docker container for radare
[2021-06-02 12:41:38][frontend][INFO]: Pulling pdf report container

Pulling PDF report container has been stopped in this step. I don't know how to operate it next?

13724460607, 2021-06-02:
barkos@barkos:~$ cat install.log 
[2021-06-02 12:38:40][install][WARNING]: Could not update python package pyyaml. Was not installed using pip originally
[2021-06-02 12:41:11][install][WARNING]: Could not update python package pyyaml. Was not installed using pip originally
[2021-06-02 12:41:11][common][WARNING]: variety spec not overwritten

jstucke, 2021-06-02:
what happens when you try to pull the PDF report container manually?

```
docker pull fkiecad/fact_pdf_report:latest
```

closed 2021-06-03 by 13724460607