fkie-cad/FACT_core #588: how to just get /rest/file_object/uid/cwe_checker infomartion?

logerrors, 2021-06-02:
hi, i have a problem, if your get /rest/file_object/uid , it will get very huge binwalk information, but i don't need it. just want the cwe_checker.

jstucke, 2021-06-07:
Hi, there is no functionality for this at the moment. You could filter the output yourself, though. If you have installed the tool `jq` you could for example do

```
curl -X GET "http://localhost:5000/rest/file_object/<uid>" | jq .file_object.analysis.cwe_checker
```

Alternatively, you could directly query the MongoDB database:
```
db.getCollection('file_objects').find({"_id": <uid>}, {"processed_analysis.cwe_checker": 1})
```

rhelmke, 2021-11-12:
Hey, we are going to close this issue for the time being. In case there are any followup questions regarding this matter, feel free to re-open it.

Cheers.

closed 2021-11-12 by rhelmke