fkie-cad/FACT_core #589: Scanning progress is stuck

weekbin, 2021-06-04:
when analysising firmware, scaning progress may stuck for long time cause of cwe plugin.

Is there any way to improve performance or set timeout to shut it off.

Or may the result of cwe plugin can output part by part ?

[image]

jstucke, 2021-06-07:
The analysis of the cwe_checker may take some time for large executables.

To improve the analysis speed there are multiple options:
- running more analysis processes in parallel (adjustable through the config file `config/main.cfg` by increasing the number of threads for the cwe_checker)
- lowering the timeout (`TIMEOUT_IN_SECONDS`) of the analysis in `src/plugins/analysis/cwe_checker/code/cwe_checker.py` (you may not get results then, though)
- only run the plugin manually on the files that are relevant or deactivating the plugin if you don't need it

rhelmke, 2021-11-12:
Hey, we are going to close this issue for the time being, as there is few activity in here. In case there are any followup questions regarding this matter, feel free to re-open it.

Cheers.

closed 2021-11-12 by rhelmke