fkie-cad/FACT_core #590: When I generated the PDF report, there was no detailed report, just a list of data;

13724460607, 2021-06-07:
I think there is a detailed description on the PDF report. How should I operate it?Is it my program?
[image]

jstucke, 2021-06-07:
I'm not sure if I understand your question correctly. This is the way the PDF report is supposed to look. It contains a brief summary of the most important results that were found during the analysis and nothing more.

13724460607, 2021-06-07:
Yes, if you want to see the full result, can you only access it from a Web interface or REST endpoint?

jstucke, 2021-06-07:
Yes, the PDF report does not contain all results. It is intended as a short summary.

13724460607, 2021-06-07:
Thank you. I have received your patient reply

closed 2021-06-07 by 13724460607