fkie-cad/FACT_core #615: CVE_Lookup Error

D3adP3nguin, 2021-07-29:
When running start_all_installed_facts_components I get the following error:
File "/home/d3adpenguin2/FACT_core/src/plugins/analysis/cve-lookup/code/vuln_lookup_plugin.py in line 14 in <module>
    from internal.database_interface import DB,DB_NAME,QUERIES 
    File "/usr/local/lib/python3.9/dist-packages/pluginbase.py", line 438, in plugin_import
          return self._system_import(import_name, globasl, locals, 
ModuleNotFoundError: No module named 'internal.database_interface'

Im not sure if i can do some pip install to fix the modulenotfounderror, but im not certain. Any help would be greatly appreciated.

I even tried to install the component separately using the GitHub reference(https://github.com/fkie-cad/FACT_analysis-plugin_CVE-lookup)

D3adP3nguin, 2021-07-29:
Ok so an update. Even when getting this error the section of cve-lookup gets filled up with CVE's 
is this an error message when it cant find a cve for what its looking up at the moment? or is there still something wrong ?

rhelmke, 2021-07-30:
Hey,

I'm sorry but I can not replicate this error and I'm not quite sure what happened here :-/. Multiple things are catching my attention:

1. We are aware of some import pathing issues that require us to implement a workaround. An example can be found in the [file mentioned in your backtrace](https://github.com/fkie-cad/FACT_analysis-plugin_CVE-lookup/blob/master/code/vuln_lookup_plugin.py#L13-L19). As you can see, part of said workaround is catching the import error you are encountering. Simply put: this error should never reach you.
2. Please don't use the plugin version from the [standalone repository](https://github.com/fkie-cad/FACT_analysis-plugin_CVE-lookup), as that repository [went stale](https://github.com/fkie-cad/FACT_analysis-plugin_CVE-lookup/commits/master) two years ago and development [continued](https://github.com/fkie-cad/FACT_core/commits/master?branch=master&path[]=src&path[]=plugins&path[]=analysis&path[]=cve_lookup) in the FACT_core repository. A lot has changed since then :-)

I suggest you start over again: clone a fresh copy of FACT_core and install it using our install guide. The `cve_lookup` plugin that ships with the default installation should work just fine.

D3adP3nguin, 2021-07-30:
So I noticed even after getting the Error message information still propagated in the `cve-lookup` section. So it is working but there is just some odd error being thrown. 
Thank you for the timely response and Ill be sure not to use the standalone plugins.

jstucke, 2021-08-02:
Hi, which version of FACT are you using? The latest master branch or the release version? 
Could you post the other error that occurred?

D3adP3nguin, 2021-08-11:
So it looked like even when the error was being thrown the cve-lookup field was being filled up with information. I was using the latest release version I believe. 
I dont have notes on the other error must have figured out a fix for it my apologies

rhelmke, 2021-08-12:
Hey,

I consider this issue as solved for the time being. Closing :-)

closed 2021-08-12 by rhelmke