fkie-cad/FACT_core #625: bootstrap-editable is no longer maintained

mistressofjellyfish, 2021-08-19:
A dependency, https://github.com/vitalets/x-editable, is no longer maintained. The library seems to be used in exactly one place, `src/web_interface/templates/user_management/edit_user.html`; but as far as I can tell, I am pretty sure it isn't compatible with bootstrap 4 anyways.

In any case, please consider to remove it.

jstucke, 2021-08-20:
It is used for editing the roles of users in FACT (only visible when authentication is enabled in `main.cfg`). Though the styling seems to be a bit broken (because of bootstrap 4), the functionality seems to still work fine. 
Is there a concrete reason to remove it? We would need to build a new interface for editing the user roles.

mistressofjellyfish, 2021-08-20:
The artifact used is from 2013, Bugs have since been fixed, and the library is (according to the readme) "not actively maintained".

Since the whole frontend tooling is so old, it is hard to find a good replacement that is maintained. So basically there are two options going forward:

(a) fork x-editable and merge [the 40 or so open pull requests](https://github.com/vitalets/x-editable/pulls) that fix - besides other things - security bugs, out-of-date dependencies, and bugs with newer browsers.
(b) remove x-editable, and replace it with a few lines of vuejs. I've had a look at it, and it doesn't seem to be too complicated - 150-200 LoC tops; fact doesn't seem to be anywhere nearly setup to really take advantage of vue. Just make sure to use vue3, because vue2 is just about to be "the old version".

That being said, vue could also be used to nuke the 5 years out of date angular dependency that might be affected by a CVE. Full disclosure: I have a greater issue with the installation process and the dependency management, but haven't sorted my thoughts yet on how the situation could be improved. My point isn't about ranting but providing a possible way to solve the problems I encounter.

PR #628 (2021-09-08): Remove x editable
resolves #625 

- removed x-editable dependency
- added modal for changing user roles
closed 2021-09-08 by jstucke