fkie-cad/FACT_core #634: Outdated dependency: Yara

mistressofjellyfish, 2021-08-28:
FACT downloads and builds Yara 3.7.1. This version is outdated, security related bugs (memory leaks, buffer overruns, the like) have since been fixed.

both debian buster and bullseye ship newer versions (4.0.5~something), so I'm not entirely sure why the installer builds an outdated version from source.

Please consider removing the source build in favor of distribution-provided packages. If they are too old, it should be the users concern to provide a sufficiently recent version.

0xricksanchez, 2021-09-01:
As far as I'm informed, that pinned yara version 3.7.0 is due to it being the latest version that is being served from the Ubuntu bionic repositories (18.04). Even Ubuntu 20.04 (focal) only offers yara 3.9 from its package repositories. 

Making the user having to provide N number of up-to-date packages (looking at yara not being the only case where this could apply) seems like an unreasonable approach. Version clashes due to local installation of some packages is annoying, so eventually the dockerized version of FACT should hopefully mitigate against such scenarios :). 

I'll discuss with the others if building from source is an option, or if  dropping major support for Ubuntu 18.04 short-term would already solve most of this problem .

mistressofjellyfish, 2021-09-01:
> As far as I'm informed, that pinned yara version 3.7.0 is due to it being the latest version that is being served from the Ubuntu bionic repositories (18.04).

FACT downloads a source tarball and builds it from source, if the version used is not on the system. So... why not pin it to the latest version? There is no reason to assume a newer version won't compile on 18.04.

> Making the user having to provide N number of up-to-date packages (looking at yara not being the only case where this could apply) seems like an unreasonable approach.

3.7.1 is not up-to-date. I'm not advocating to use the alwayst-latest-version, but a not-pinned one. Versions from the repository have at least a chance to get bug fixes (that are not features). Upstream doesn't backport. Debian does (sometimes).

"Making the user provide N number of up-to-date packages" is exactly what a manual installation _should_ do. I'm now like 98% through the installation process / looking at the code, and all FACT does with yara is wrapping its command line. So, assuming that this doesn't change between major versions, FACT should be fine. And, in all reality, tests are there. Just spawn another CI container and run the tests there to see if they work or not. 

I know that FACT doesn't consider itself to be hardend against attacks. And as far as the UI is concerned, I'm fine with that. But this is involved with handling untrusted files from untrusted sources. The whole installer structure introduces so much maintenance burden on your end to just keep things somewhat up-to-date that I would argue that it's a battle against windmills, and most of the assumptions the installer makes when installing are so deeply flawed anyways that it can't be repaired easily.

Don't get me wrong: that's entirely fixable. But it needs a _serious_ shift in installer philosphy, which inherently puts a bit more of the installation workload on the user. But on the flipside, it would make deployment in docker containers or environments other than some debian derivates _much_ easier. Because the number one reason why I have so much headache installing FACT on my system is that I don't run a debian based distribution, which renders the current installer completely useless.

mistressofjellyfish, 2021-09-02:
On another note: the plugin "ip_and_uri_finder" installs `yara-python >= 3.5` as a dependency. yara-python is distributed as a wheel, with a python extension that statically links yara.

This means that:
- parts of FACT may use a different version of yara than 3.7.1 anways
- this version might be _lower_ than 3.7.1
- but in reality would be the latest version at time of installation (4.1.2)

So we're version mixing already. Why not skip the homebrew yara wrapper entirely and just use yara-python? This would retain the ability to OS-independently update, while at the same time reducing the maintainance burden on your end.

jstucke, 2021-09-06:
The version was pinned at some point because there were problems with the newest version. A lot has happened since and it shouldn't be a problem to use a newer (the newest?) version at this point. We will try it and fix it if it works.

mistressofjellyfish, 2021-09-06:
I would still recommend to move Code that uses shell execution to invoke yara to yara-python, and then fix its version if the newest causes problems. This makes installation and dependency management much easier.

jstucke, 2021-09-07:
Sadly, we cannot fully replace every call to yara with yara-python, because the recursive search ("binary search") would be much slower (and it can take hours on a large database already).

mistressofjellyfish, 2021-09-07:
I see, cli-yara implements threading (albeit with a maximum of 32 threads). Obviously, python doesn't really scale there. In that case – would it be possible to update the documentation to mention this? 

This is of course a dealbreaker for any implementation that directly interfaces with python in this way (and also not one that has an easy way out to a nice solution; that is a nicer solution than to script the CLI).

But in the end, this is still a dependency management problem, even if yara-python is not the way to solve it.