fkie-cad/FACT_core #635: checksec.sh is installed regardless of packaged version

mistressofjellyfish, 2021-08-28:
https://github.com/fkie-cad/FACT_core/blob/1aa7de2205f788a128274ec4ebb6be80825cc556/src/install/backend.py#L144-L155

Debian stable (bullseye) ships the latest release, checksec.sh 2.4.0. My distribution of choice does the same. Is there any reason to ignore the distribution-provided packaged versions on these platforms outright? If a sufficiently recent version (judging from when the code was added, 2.4.0 should suffice) is present, why not use the packaged version?

In any case, checksec.sh goes to great length to ensure that its internal auto-update only updates to a signed release (providing a pubkey and checking the signature of checksec.sig against the downloaded checksec script). FACT completely ignores that.

0xricksanchez, 2021-09-01:
Similar to my answer in #634 this likely stems from the checksec.sh not being part of Ubuntu 18.04 package repositories. I'm aware that it seems to be packaged in newer distributions, like Ubuntu 20.04, where a manual installation from git would not be necessary anymore.

jstucke, 2021-09-06:
Not all distributions that we support come with a packaged version of checksec.sh (i.e. Ubuntu 18.04) and we want to avoid separate installations for each distribution when possible (as all of them need to be maintained). The internal update mechanism does not seem to help us with the initial install but we could check whether we can utilize it to update the installed version.