fkie-cad/FACT_core #636: Outdated dependency: binwalk

mistressofjellyfish, 2021-09-01:
The FACT plugin binwalk installs binwalk from a fork: https://github.com/jstucke/binwalk/commit/1534001b96b8d543dcbb52845526326b61119f8c

Since this version has been forked, binwalk ditched [PyCrypto](https://github.com/ReFirmLabs/binwalk/issues/519) as one of its dependencies, as that is unmaintained and affected by serveral CVEs.

PyCrypto is in the process of [being pulled from pypi](https://github.com/pycrypto/pycrypto/issues/301), so this version might not be installable in the near future.

The commit in question has [since been merged](https://github.com/ReFirmLabs/binwalk/commits/master/src/binwalk/modules/entropy.py) and [released](https://github.com/ReFirmLabs/binwalk/releases/tag/v2.3.0).

FACT should require binwalk >= 2.3.0 present, and ditch the manual installing in this way.

mistressofjellyfish, 2021-09-01:
As a sidenote, debian stable ships 2.3.1, same as Ubuntu 21.04. Latest LTS 20.04 still ships 2.2.0.

0xricksanchez, 2021-09-01:
Reverting to the ReFirmlabs repository for binwalk now that multiples relevant patches have been merged looks like a sane idea.

jstucke, 2021-09-06:
Resolved in #637

closed 2021-09-06 by jstucke