fkie-cad/FACT_core #675: Feature Request: Support Device Tree Expansion

frakman1, 2021-10-31:
Most firmware comes bundled with a device-tree.
It would be ideal if FACT could expand the device-tree to help inpect the firmware better.

[image]

It currently stops processing at the .dtb. Allowing the user to drill down to the contents would be very useful additional functionality.

I can view its contents manually in linux using the syntax:
`dtc -I dtb -O dts -o - filename.dtb`

At the very least, this output could be part of the 'Show Preview' view at the bottom or one of the views you can select from the left

jstucke, 2021-11-02:
Actually someone already wrote a plugin for finding and extracting device trees: https://github.com/fabiansteiner/dtb_finder
Theoretically, you could simply clone it and put it in your plugins folder. I don't know if it still works and if it does what you are looking for, though. 

Also, if its only e.g. conversion to .dts, it shouldn't be too complicated to write your own plugin. There is even a guide here: https://github.com/fkie-cad/FACT_core/wiki/analysis-plugin-development. It may be a bit complicated inside a docker container, though. But theoretically, it should work to mount the plugin's folder to the analysis plugin folder. We also added a custom MIME type for this that could be used as "whitelist": `linux/device-tree` (see https://github.com/fkie-cad/fact_helper_file/blob/master/fact_helper_file/mime/custom_mime_linux).

We will probably also find some time to do it at some point (I can't make guarantees as to when that will be, though).

frakman1, 2021-11-02:
I looked at that project and tried it. I don't know if I am installing it correctly though.
I entered the `fact` container and cloned the repo into the `analysis` folder:

`fact@Kubuntu18:/opt/FACT_core/src/plugins/analysis$ git clone https://github.com/fabiansteiner/dtb_finder.git`

I then went into the `/opt/FACT_core/src` folder and ran `./install.py -B` but the logs don't indicate that the `dtb_finder` was installed:

```bash
fact@Kubuntu18:/opt/FACT_core/src$ ./install.py -B
[2021-11-02 14:25:31][install][INFO]: FACT Installer 1.2
[2021-11-02 14:25:31][common][INFO]: Updating system
[2021-11-02 14:25:35][install][INFO]: Installing apt-transport-https autoconf automake build-essential git libtool python3 python3-dev unzip libfuzzy-dev libmagic-dev
[2021-11-02 14:25:37][common][INFO]: Installing python3 pip
[2021-11-02 14:26:30][install][WARNING]: Pip package pyyaml is already installed with distutils. This may Cause problems:
ERROR: Cannot uninstall 'PyYAML'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.

[2021-11-02 14:26:46][common][WARNING]: variety spec not overwritten
[2021-11-02 14:26:46][install][INFO]: Installing libjpeg-dev libssl-dev binutils file openssl bison flex pkg-config
[2021-11-02 14:27:12][backend][INFO]: Skipping yara installation: Already installed and up to date
[2021-11-02 14:27:12][backend][INFO]: Installing checksec.sh
[2021-11-02 14:27:12][backend][INFO]: Pulling fact extraction container
[2021-11-02 14:27:13][backend][INFO]: Installing oms plugin.
[2021-11-02 14:27:33][backend][INFO]: Finished installing oms plugin.

[2021-11-02 14:27:33][backend][INFO]: Installing crypto_hints plugin.
[2021-11-02 14:27:34][backend][INFO]: Finished installing crypto_hints plugin.

[2021-11-02 14:27:34][backend][INFO]: Installing software_components plugin.
[2021-11-02 14:27:34][backend][INFO]: Finished installing software_components plugin.

[2021-11-02 14:27:34][backend][INFO]: Installing cve_lookup plugin.
[2021-11-02 14:29:10][backend][INFO]: Finished installing cve_lookup plugin.

[2021-11-02 14:29:10][backend][INFO]: Installing ip_and_uri_finder plugin.
[2021-11-02 14:29:17][backend][INFO]: Finished installing ip_and_uri_finder plugin.

[2021-11-02 14:29:17][backend][INFO]: Installing binwalk plugin.
[2021-11-02 14:29:33][backend][INFO]: Finished installing binwalk plugin.

[2021-11-02 14:29:33][backend][INFO]: Installing kernel_config plugin.
[2021-11-02 14:29:39][backend][INFO]: Finished installing kernel_config plugin.

[2021-11-02 14:29:39][backend][INFO]: Installing qemu_exec plugin.
[2021-11-02 14:29:40][backend][INFO]: Finished installing qemu_exec plugin.

[2021-11-02 14:29:40][backend][INFO]: Installing file_system_metadata plugin.
[2021-11-02 14:29:41][backend][INFO]: Finished installing file_system_metadata plugin.

[2021-11-02 14:29:41][backend][INFO]: Installing input_vectors plugin.
[2021-11-02 14:29:42][backend][INFO]: Finished installing input_vectors plugin.

[2021-11-02 14:29:42][backend][INFO]: Installing cwe_checker plugin.
[2021-11-02 14:29:47][backend][INFO]: Finished installing cwe_checker plugin.

[2021-11-02 14:29:47][backend][INFO]: Installing users_and_passwords plugin.
[2021-11-02 14:31:41][backend][INFO]: Finished installing users_and_passwords plugin.

[2021-11-02 14:31:41][backend][INFO]: Installing linter plugin.
[2021-11-02 14:31:54][backend][INFO]: Finished installing linter plugin.

[2021-11-02 14:31:54][backend][INFO]: set environment variables...
[2021-11-02 14:31:54][backend][INFO]: Creating firmware directory
Create signature directory /opt/FACT_core/src/analysis/signatures
Compile signatures in /opt/FACT_core/src/plugins/analysis/crypto_hints/signatures
Compile signatures in /opt/FACT_core/src/plugins/analysis/software_components/signatures
Compile signatures in /opt/FACT_core/src/plugins/analysis/known_vulnerabilities/signatures
Compile signatures in /opt/FACT_core/src/plugins/analysis/crypto_material/signatures
Compile signatures in /opt/FACT_core/src/plugins/analysis/dtb_finder/signatures
[2021-11-02 14:31:54][install][INFO]: installation complete
[2021-11-02 14:31:54][install][WARNING]: If FACT does not start, reload the environment variables with: source /etc/profile
[... 43 lines cut ...]
[pid: 350|app: 0|req: 216/1053] 192.168.86.31 () {36 vars in 932 bytes} [Tue Nov  2 15:16:08 2021] GET /static/Pacman.gif => generated 0 bytes in 4 msecs (HTTP/1.1 304) 5 headers in 285 bytes (0 switches on core 1)
Process ExceptionSafeProcess-15:4:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/opt/FACT_core/src/helperFunctions/process.py", line 57, in run
    raise exception
  File "/opt/FACT_core/src/helperFunctions/process.py", line 52, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/FACT_core/src/analysis/PluginBase.py", line 123, in process_next_object
    finished_task = self.analyze_file(task)
  File "/opt/FACT_core/src/analysis/PluginBase.py", line 69, in analyze_file
    fo = self.process_object(file_object)
  File "/opt/FACT_core/src/plugins/analysis/dtb_finder/code/dtb_finder.py", line 79, in process_object
    parentAnalysis = parent.processed_analysis[self.NAME]
KeyError: 'dtb_finder'
[... cut ...]
[pid: 348|app: 0|req: 222/1054] 192.168.86.31 () {36 vars in 1129 bytes} [Tue Nov  2 15:16:08 2021] GET /ajax_root/f0ff064c91b9ba52d75c2364b6fa30445e99f9725cb62156b161fc536454c660_33623/6e17d0339e0218aeb5b265fee08b4d290f2f4810ae6ed0ddd40d4c7453fee96b_22413312 => generated 570 bytes in 23 msecs (HTTP/1.1 200) 4 headers in 216 bytes (1 switches on core 0)
[2021-11-02 15:16:08][Analysis][INFO]: Analysis Completed:
UID: f0ff064c91b9ba52d75c2364b6fa30445e99f9725cb62156b161fc536454c660_33623
```

I'm not sure what else to try at this point. If you have any ideas, I can try them. If not, I can open a ticket on the `dtb_finder` github project page.

jstucke, 2021-11-02:
Actually the plugin installation process changed since this plugin was implemented (e.g. from install.sh to install.py), that's why `install.py -B` won't work. It may still possible to run the install.sh in the plugin folder individually, though.

frakman1, 2021-11-02:
Thanks. I ran the `install.sh` script within the `dtb_finder` folder and it seems that all it does is install the `device-tree-compiler` dependency.  

I double-checked the output of `install.py -B` and it looks like it had some `dtb_finder` log at the end: 
`Compile signatures in /opt/FACT_core/src/plugins/analysis/dtb_finder/signatures`

jstucke, 2021-11-03:
Does the plugin still work and does it actually do what you are looking for?

frakman1, 2021-11-03:
Unfortunately, I still get the error mentioned above: `KeyError: 'dtb_finder'`

frakman1, 2021-11-11:
Do you have any recommendations for things I can try within the working fact container?
The module appears in the list of tests but throws errors when running.

The folder looks like this:
```bash
fact@Kubuntu18:/opt/FACT_core/src/plugins/analysis/dtb_finder$ tree
.
|-- README.md
|-- __init__.py
|-- code
|   |-- __init__.py
|   |-- __pycache__
|   |   `-- dtb_finder.cpython-38.pyc
|   `-- dtb_finder.py
|-- install.sh
|-- restaccess
|   `-- get_dtbs.py
|-- signatures
|   `-- devicetree.yara
|-- test_files
|   |-- DeviceTree.zip
|   `-- ddwrt-linksys-wrt1900acs-webflash.bin
`-- view
    `-- dtb_finder.html
```

jstucke, 2021-11-12:
We are currently working on a simple plugin that does the conversion from dtb to dts. https://github.com/fkie-cad/fact_extractor/pull/88 already added unpacking of device tree blobs from u-boot images to the extractor.

frakman1, 2021-11-12:
May I ask what is the preferred method for troubleshooting/viewing logs? 
- `docker logs --follow fact`
- viewing `/var/log/fact/main_backend.log`
- WebUI->Admin->Logs

or something else?

jstucke, 2021-11-12:
"WebUI->Admin->Logs" and `main_{backend,frontend}.log` should have the same content. `docker logs --follow fact` should in addition to that also have entries with log level "info".

frakman1, 2021-11-12:
> We are currently working on a simple plugin that does the conversion from dtb to dts. [fkie-cad/fact_extractor#88](https://github.com/fkie-cad/fact_extractor/pull/88) already added unpacking of device tree blobs from u-boot images to the extractor.

Will doing `git pull` from within the container bring this functionality in? Do I have to do anything else to get it to work?

jstucke, 2021-11-15:
The extractor does not run inside the container. It has its own container (fkiecad/fact_extractor). The only problem is that it does no longer get built automatically since the changes to Docker Hub.  We are currently trying to find a solution (probably with a GitHub action), but you could also build the image yourself to get the latest version (https://github.com/fkie-cad/fact_extractor). If you use a different name/tag then you need to replace the other extractor image in the code of FACT_core inside your container in `FACT_core/src/unpacker/unpack_base.py`.

jstucke, 2021-11-15:
It looks like the github action already works, so a `docker pull fkiecad/fact_extractor:latest` should suffice.

frakman1, 2021-11-15:
OK, I pulled it fromwithin the fact container.

Now, how do I update an analysis that includes a .dtb file since it there's no pluging for dtb and I don't see an unpacker option in the list of plugins when performing an update-analysis.
Complete redo-analysis from scratch takes a few hours which I was hoping to avoid.

frakman1, 2021-11-15:
I tried uploading a .dtb file only as a test but saw this error in the docker logs:

```bash
[2021-11-15 13:31:45][unpack_base][ERROR]: Failed to execute docker extractor with code 1:
WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
Traceback (most recent call last):
  File "./docker_extraction.py", line 68, in <module>
    sys.exit(main(_parse_args()))
  File "./docker_extraction.py", line 56, in main
    input_file = list(input_dir.iterdir())[0]
  File "/usr/lib/python3.8/pathlib.py", line 1122, in iterdir
    for name in self._accessor.listdir(self):
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/extractor/input'

Process Unpacking-Worker-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/opt/FACT_core/src/helperFunctions/process.py", line 57, in run
    raise exception
  File "/opt/FACT_core/src/helperFunctions/process.py", line 52, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 69, in unpack_worker
    extracted_objects = unpacker.unpack(fo)
  File "/opt/FACT_core/src/unpacker/unpack.py", line 40, in unpack
    extracted_files = self.extract_files_from_file(file_path, tmp_dir.name)
  File "/opt/FACT_core/src/unpacker/unpack_base.py", line 31, in extract_files_from_file
    raise RuntimeError(error)
RuntimeError: Failed to execute docker extractor with code 1:
WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
Traceback (most recent call last):
  File "./docker_extraction.py", line 68, in <module>
    sys.exit(main(_parse_args()))
  File "./docker_extraction.py", line 56, in main
    input_file = list(input_dir.iterdir())[0]
  File "/usr/lib/python3.8/pathlib.py", line 1122, in iterdir
    for name in self._accessor.listdir(self):
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/extractor/input'

[2021-11-15 13:31:45][process][ERROR]: Exception in Unpacking process:
Traceback (most recent call last):
  File "/opt/FACT_core/src/helperFunctions/process.py", line 52, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 69, in unpack_worker
    extracted_objects = unpacker.unpack(fo)
  File "/opt/FACT_core/src/unpacker/unpack.py", line 40, in unpack
    extracted_files = self.extract_files_from_file(file_path, tmp_dir.name)
  File "/opt/FACT_core/src/unpacker/unpack_base.py", line 31, in extract_files_from_file
    raise RuntimeError(error)
RuntimeError: Failed to execute docker extractor with code 1:
WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
Traceback (most recent call last):
  File "./docker_extraction.py", line 68, in <module>
    sys.exit(main(_parse_args()))
  File "./docker_extraction.py", line 56, in main
    input_file = list(input_dir.iterdir())[0]
  File "/usr/lib/python3.8/pathlib.py", line 1122, in iterdir
    for name in self._accessor.listdir(self):
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/extractor/input'

```

I created and ensured that  `/tmp/extractor/input` was in the `fact` container

frakman1, 2021-11-15:
It seems this error is occurring no matter what firmware I upload and try.

jstucke, 2021-11-16:
I was able to reproduce this problem but I'm not sure why it occurred in the first place. It seems something undid one of the FACT docker patches to the FACT core configuration (see https://github.com/fkie-cad/FACT_docker/blob/main/0001_main_cfg.patch). To be more precise, the path for temporary files was changed back to `/tmp`. The unpacking worked again after I changed the entry `temp_dir_path` back to `/tmp/fact-docker-tmp` inside the docker container (`/opt/FACT_core/src/config/main.cfg`).

jstucke, 2021-11-16:
Is seems what I wrote was not clear. The extractor now unpacks `.dtb` files from u-boot images (the dtb files themselves are not unpacked). The analysis plugin which converts binary device tree blobs to dts is not yet merged into FACT_core. The [PR](https://github.com/fkie-cad/FACT_core/pull/693) was recently opened, though.

frakman1, 2021-11-16:
I see. Thank you for the clarification.
I look forward to testing this feature once it is integrated.

jstucke, 2021-12-02:
#693 was merged recently and adds a plugin that converts binary device tree blobs into text and saves the result. You could try it out and test if it does what you were looking for.

We will probably extend the plugin to use `fdtdump --scan` to also scan arbitrary files for device trees (currently it only works on dtb files).

frakman1, 2021-12-02:
Thank you for the note. Is it possible to just do  a `git pull` within the fact docker container for this to work or do I need to run a top level installation script as well?

frakman1, 2021-12-03:
I can confirm that this now works for me. Thank you so much!

closed 2021-12-03 by frakman1

jstucke, 2021-12-10:
#711 extended the plugin so that it also searches for file trees in arbitrary files using `fdtdump` and is already merged into master.