fkie-cad/FACT_core #676: What is the function of 8000 port?

barkosHuang, 2021-11-02:
What is the function of 8000 port?

I installed mobsf on my server. It uses 8000 port. I saw in fact that 8000 port also exists here. Can I directly modify 8000 port to other ports in the fact program?
[image]

jstucke, 2021-11-02:
It is used for radare (the "View in radare" button on the analysis results page). It should be safe to change it, but you need to change the port in `_get_radare_endpoint()` in `src/web_interface/components/io_routes.py` (and possibly other locations) if you still want to use that button.

Also you could simply turn off the radare containers by commenting out the `docker-compose` lines in `src/start_fact_frontend.py` if you don't need them.

rhelmke, 2021-11-12:
Hey, we are going to close this issue for the time being. In case there are any followup questions regarding this matter, feel free to re-open it.

Cheers.

closed 2021-11-12 by rhelmke