fkie-cad/FACT_core #679: openssl version detection logic is incorrect

frakman1, 2021-11-02:
When looking at the qemu output for openssl, I see that it tries to detect the version by providing the `--version` parameter to the `openssl` command. 

[image]

Side note: Not sure why the output is not formatted correctly (like the `no-parameter` output)

Openssl's version can be retreived by calling `openssl version`. For example:

```bash
$ openssl version
OpenSSL 1.1.1  11 Sep 2018
```

It would be good if the qemu plugin had logic to try `version`  to detect binaries like openssl.

jstucke, 2021-11-03:
The "qemu exec" plugin is not really intended to detect software versions and I'm somewhat hesitant to add more parameters ("version" without dashes is also somewhat exotic), since these parameters are tried for every executable (and the runtime of the plugin is quite high as it is). The idea behind the qemu plugin was to check if the executables in the file system can be emulated without problems. 

The "software components" plugin should detect OpenSSL including its version, though. Did it find the correct version in your case?

frakman1, 2021-11-03:
> The "software components" plugin should detect OpenSSL including its version, though. Did it find the correct version in your case?

Sort of. It detects the httpd binary as openssl:
[image]

It detects the libcrypto.so.1.1 library as openssl with the correct version 1.1.1b
[image]

But it does not list the actual `openssl` binary for some reason, nor the `libssl` library. Weird.

Anyway, sometimes the only insight into what version of code is running is the output from the qemu analysis page. This is because the automatic software detection logic is quite poor. Out of hundreds of linux modules, only a small number is detected like Linux Kernel, Bash, OpenSSL and a few others.

I can understand that adding this option to each check may introduce delay, but perhaps it can be selectively applied to known modules that use this syntax.

jstucke, 2021-11-05:
> But it does not list the actual openssl binary for some reason, nor the libssl library. Weird.

Most of the signatures are based on regex matching, which has obvious limitations. Other binaries or libraries may match if they are linked against OpenSSL (could be the case here for httpd). 

> Anyway, sometimes the only insight into what version of code is running is the output from the qemu analysis page. This is because the automatic software detection logic is quite poor. Out of [...]

Feel free to contribute to the software signatures if you think that there are important ones missing. You can find the signatures [here](https://github.com/fkie-cad/FACT_core/tree/master/src/plugins/analysis/software_components/signatures).

> I can understand that adding this option to each check may introduce delay, but perhaps it can be selectively applied to known modules that use this syntax.

We can probably deal with one additional argument to be checked. A set of special cases would be hard to maintain and would probably feel very incomplete.

rhelmke, 2021-12-03:
Closing. As @jstucke said, we are grateful for any kind of contribution to this project. Feel free to create a pull request with new software signatures :-).

closed 2021-12-03 by rhelmke