fkie-cad/FACT_core #685: I rebuild fact_pdf_report docker,But report function can't work

Key20, 2021-11-06:
I rebuild the docker(https://github.com/fkie-cad/fact_pdf_report) AND replace the old docker image(fact_pdf_report),BUT the report function didn't work.I don't know why.Could you give me a help?Thank you.

jstucke, 2021-11-08:
Were there any error messages in the logs? If running the container fails, the error should be logged to the command line.

You could also try to run the new image with the existing tests for the pdf report in `test/unit/helperFunctions/test_pdf.py` (e.g.  `pytest test/unit/helperFunctions/test_pdf.py::test_build_pdf_report`) or modify them if necessary. This should be helpful for debugging.

rhelmke, 2022-09-15:
The issuer didn't follow up on this one. Closing.

closed 2022-09-15 by rhelmke