fkie-cad/FACT_core #691: cwe_checker causes backend offline.

AlexPeiran, 2021-11-12:
Last month, the plugin cwe_checker worked well and help me a lot.To make sure the plugin is up-to-date, I re-install Fact regularly.  
Recently I found out that cwe_checker does not work properly and causes backend offline. 

The log shows:
[2021-11-11 11:45:04][cwe_checker][ERROR]: cwe_checker execution failed: Execution of Ghidra plugin failed: Process was terminated.
INFO  REPORT: Save succeeded for file: /input (HeadlessAnalyzer)  
INFO  REPORT: Post-analysis succeeded for file: /input (HeadlessAnalyzer)  
………………

UID: cd5eabea0aa4c0aef0c6cb571233b91f5e6e2f73e126ffa57776ce28518454e8_215240
[2021-11-11 12:58:37][cwe_checker][ERROR]: Timeout or error during cwe_checker execution.
UID: 95d80c685435bcc979c3ef04252fc7c471eee442c305520eede47ee882e8d5ad_2840360
[2021-11-11 13:01:55][docker][WARNING]: [Docker]: encountered process error while processing
[2021-11-11 13:02:06][cwe_checker][ERROR]: Timeout or error during cwe_checker execution.
………………

Should I try to install an older version of cwe_checker?

Enkelmann, 2021-11-12:
That the cwe_checker plugin throws errors (like those in your log) for some files is normal, but that should not cause problems with the whole backend of FACT. Could you describe what you mean with it causing "backend offline"?

Regarding the version of the cwe_checker: As far as I remember, the standard version of the cwe_checker installed with FACT is its latest stable version (v0.5). At the moment I recommend sticking to that version, since newer dev-versions of the cwe_checker currently have some runtime issues, which causes the cwe_checker-FACT-plugin to run into timeouts quite often.

AlexPeiran, 2021-11-12:
In the web interface , I can see the status of the "backend" is "offline", just in the system dashboard. And the analysis process stopped. I can see only 1 process of cwe-checker is running(sometimes None) after a few time since the analysis started.

When I use the command "docker image list", I can see that cwe-checker has been updated a few days ago（the old one's tag is "<None>", the newer one's tag is “Latest”）. Since then, analysis has been difficult to start.

By the way, although I use FACT to analyze the firmware of a large rack-mounted router, I can guarantee that the specifications of the virtual machine running by FACT are appropriate (32U64T 128GB RAM)

Enkelmann, 2021-11-15:
I would suggest switching to the stable release of the cwe_checker then by removing the current image (`docker image rm fkiecad/cwe_checker`) and then pulling the stable docker container (`docker pull fkiecad/cwe_checker:stable`).

Because of the encapsulation through Docker, the only way (that I can think of right now) how the cwe_checker could destabilize the rest of FACT is if your RAM overflowed during the analysis. In that case switching to the stable release of the cwe_checker should solve the problem. But I would also recommend running the firmware analysis once with deactivated cwe_checker, just to double-check that the issue really is related to the cwe_checker plugin. If you are able to share the firmware image, we could also try to reproduce the issue on our end.

AlexPeiran, 2021-11-16:
Sorry, for some reason, the firmware can't be shared with you. I've done three analyses these days without cwe_checker, so I think the problem is caused by the latest cwe_checker. I will try again later this week with the stable version and send the result to you .

jstucke, 2022-12-16:
Since there is no activity I will close this issue. Please feel free to reopen the issue if the problem persists or if you have additional questions.

closed 2022-12-16 by jstucke