fkie-cad/FACT_core #694: Known vulnerabilities is none

KongLynn, 2021-11-18:
Hi，I have scanned multiple firmwares but none of their known vulnerabilities. Can you tell me the specific scan contents of the known vulnerabilities module？ Thank you very much.

dorpvom, 2021-11-19:
Right now, the `known_vulnerabilities` plugin is more a framework - containing only a minimal amount of rules - to showcase ways of using "fusion" of existing analysis results to identify deeper vulnerabilities as can be found e.g. by a single plugin.
The [wiki](https://github.com/fkie-cad/FACT_core/wiki/Known_Vulnerability_Rules) gives a (hopefully) comprehensive explanation on how to write new rules.

As of now, we have 5 rules (found [here](https://github.com/fkie-cad/FACT_core/blob/master/src/plugins/analysis/known_vulnerabilities/internal/rulebook.py) and [here](https://github.com/fkie-cad/FACT_core/blob/master/src/plugins/analysis/known_vulnerabilities/signatures/vulnerabilities.yara)):

- Heartbleed
- [D-Link Auth Bypass](http://www.devttys0.com/2013/10/reverse-engineering-a-d-link-backdoor/)
- Hardcoded WiFi Keys
- Binaries containing the string _backdoor_
- [Netgear CGI reverse shell bug](https://nvd.nist.gov/vuln/detail/CVE-2016-6277)

KongLynn, 2021-11-25:
ok，I got it.Thanks you very much！

rhelmke, 2021-11-25:
Great, closing :-)

Feel free to re-open in case there are any further questions regarding this matter.

Cheers

closed 2021-11-25 by rhelmke