fkie-cad/FACT_core #697: It lasted a particularly long time in "Pulling fact extraction container". Is this normal？

cen6667, 2021-11-25:
[image]
I tried to install FACT on ubuntu18.04 but it's stuck on the "Pulling fact extraction container" step.
Is this normal？
What should we do next？

rhelmke, 2021-11-25:
Hi! So I just went through the installation process to verify that the installation on master still works properly, which is the case.

As you can see [in this line](https://github.com/fkie-cad/FACT_core/blob/891f8449b6bf906561e37a8a88eb8c50fce5ab3a/src/install/backend.py#L71) of the backend installation, you can see that the subject step just executes `docker pull fkiecad/fact_extractor`. To further troubleshoot, I'd advise you to execute said command in the shell and check out if there are any errors.

Also, consider the following points:

* The fact extraction container is hosted at Dockerhub, which rate limits container pulls on too many requests
* The decompressed container image is ~2GB in size, which leads to a download size of roughly 1.3GB. Depending on your internet speed, this may take a while.

cen6667, 2021-11-30:
I am a beginner from China. Therefore, network problems often occur during installation.I have encountered many difficulties in the installation process. I want to ask you some simple questions.
1. What system is better to use? ( Ubuntu18, Ubuntu20 or kali )
2. Which installation method is used? ( docker or traditional installation ) 
3. Can you give me some other suggestions about anything?
This is my first contact with open source projects. I'm lucky to meet you.
Thank you very much for your reply.

rhelmke, 2021-11-30:
1. Currently, we have workers for various Debian and Ubuntu releases in our CI pipelines. Thus, we can assert that installations from `master` run properly on these systems. It's probably wise to choose the highest supported version for FACT.
2. A conventional installation deploys FACT bare metal on your system, but pulls in various docker containers for the tools it interfaces with. If you are interested in a docker installation, please check out the [FACT_docker](https://github.com/fkie-cad/FACT_docker) repository.
3. I'm also glad to meet you, but I fear some other suggestions about anything are above the scope of both this issue and its corresponding software.

I trust we were able to pinpoint the root cause of your initial problem, which appears to be network-related. Thus, I'm closing this issue for the time being.

Cheers :-)

closed 2021-11-30 by rhelmke