fkie-cad/FACT_core #700: ELF Analysis causes internal server error

timkoers, 2021-11-28:
Hi,

I'm just trying out FACT with an ELF binary and when I click on the elf analysis, I get an internal server error.

Logs:

```
[2021-11-28 13:55:16][start_all_installed_fact_components][INFO]: starting db
[2021-11-28 13:55:17][MongoMgr][INFO]: Starting local mongo database
[2021-11-28 13:55:17][MongoMgr][INFO]: Starting DB: mongod --auth --config /home/tim/FACT_core/src/config/mongod.conf --fork --logpath /tmp/fact_mongo.log
[2021-11-28 13:55:17][fact_base][INFO]: Successfully started FACT DB-Service
[2021-11-28 13:55:17][fact_base][INFO]: Successfully started FACT DB-Service
[2021-11-28 13:55:18][start_all_installed_fact_components][INFO]: starting frontend
[2021-11-28 13:55:18][start_all_installed_fact_components][INFO]: starting backend
[2021-11-28 13:55:20][cwe_checker][INFO]: Version is cwe_checker 0.5.0

[2021-11-28 13:55:20][input_vectors][INFO]: Up and running.
[2021-11-28 13:55:20][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.8/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2021-11-28 13:55:20][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.8/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
[2021-11-28 13:55:21][Analysis][INFO]: Analysis System online...
[2021-11-28 13:55:21][Analysis][INFO]: Plugins available: ['binwalk', 'cpu_architecture', 'crypto_hints', 'crypto_material', 'cve_lookup', 'cwe_checker', 'dummy_plugin_for_testing_only', 'elf_analysis', 'exploit_mitigations', 'file_hashes', 'file_system_metadata', 'file_type', 'hardware_analysis', 'init_systems', 'input_vectors', 'interesting_uris', 'ip_and_uri_finder', 'kernel_config', 'known_vulnerabilities', 'malware_scanner', 'printable_strings', 'qemu_exec', 'software_components', 'source_code_analysis', 'string_evaluator', 'tlsh', 'users_and_passwords']
[2021-11-28 13:55:21][Unpacking][INFO]: Unpacker Module online
[2021-11-28 13:55:21][Unpacking][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2021-11-28 13:55:21][compare][INFO]: Plug-ins available: ['File_Coverage', 'File_Header', 'Software']
[2021-11-28 13:55:21][Compare][INFO]: Compare Scheduler online...
[2021-11-28 13:55:21][binary_service][INFO]: binary service online
[2021-11-28 13:55:21][binary_service][INFO]: binary service online
[2021-11-28 13:55:21][back_end_binding][INFO]: InterCom started
[2021-11-28 13:55:21][binary_service][INFO]: binary service online
[2021-11-28 13:55:22][fact_base][INFO]: Successfully started FACT Frontend
[uWSGI] getting INI configuration from /home/tim/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.20 (64bit) on [Sun Nov 28 13:55:22 2021] ***
compiled with version: 9.3.0 on 27 November 2021 14:50:03
os: Linux-5.10.16.3-microsoft-standard-WSL2 #1 SMP Fri Apr 2 22:23:49 UTC 2021
nodename: TheBeast
machine: x86_64
clock source: unix
detected number of CPU cores: 24
current working directory: /home/tim/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
your processes number limit is 63815
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address 127.0.0.1:5000 fd 3
Python version: 3.8.10 (default, Sep 28 2021, 16:10:42)  [GCC 9.3.0]
Python main interpreter initialized at 0x55b372dffd40
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500064 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
[2021-11-28 13:55:23][frontend_main][INFO]: Web front end online
WSGI app 0 (mountpoint='') ready in 1 seconds on interpreter 0x55b372dffd40 pid: 19396 (default app)
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 19396)
spawned uWSGI worker 1 (pid: 19435, cores: 2)
spawned uWSGI worker 2 (pid: 19437, cores: 2)
spawned uWSGI worker 3 (pid: 19439, cores: 2)
spawned uWSGI worker 4 (pid: 19441, cores: 2)
spawned uWSGI worker 5 (pid: 19443, cores: 2)
*** Stats server enabled on 127.0.0.1:9191 fd: 18 ***
[2021-11-28 13:55:33][update][INFO]: time to create stats: 0s
[pid: 19435|app: 0|req: 1/1] 127.0.0.1 () {44 vars in 991 bytes} [Sun Nov 28 13:55:33 2021] GET / => generated 26935 bytes in 99 msecs (HTTP/1.1 200) 2 headers in 82 bytes (1 switches on core 0)
[pid: 19435|app: 0|req: 2/2] 127.0.0.1 () {40 vars in 752 bytes} [Sun Nov 28 13:55:33 2021] GET /static/web_css/bootstrap.min.css => generated 155758 bytes in 1 msecs via sendfile() (HTTP/1.1 200) 6 headers in 247 bytes (0 switches on core 1)
[pid: 19437|app: 0|req: 2/3] 127.0.0.1 () {40 vars in 764 bytes} [Sun Nov 28 13:55:33 2021] GET /static/web_css/fontawesome/css/all.css => generated 72670 bytes in 17 msecs via sendfile() (HTTP/1.1 200) 6 headers in 236 bytes (0 switches on core 1)
[pid: 19437|app: 0|req: 2/4] 127.0.0.1 () {40 vars in 728 bytes} [Sun Nov 28 13:55:33 2021] GET /static/web_js/jquery.min.js => generated 97403 bytes in 19 msecs via sendfile() (HTTP/1.1 200) 6 headers in 256 bytes (0 switches on core 0)
[pid: 19441|app: 0|req: 2/6] 127.0.0.1 () {40 vars in 788 bytes} [Sun Nov 28 13:55:33 2021] GET /static/web_css/bootstrap-datepicker.standalone.css => generated 18056 bytes in 17 msecs via sendfile() (HTTP/1.1 200) 6 headers in 264 bytes (0 switches on core 0)
[... 108 lines cut ...]
    self.environment.handle_exception()
  File "/usr/local/lib/python3.8/dist-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/home/tim/FACT_core/src/web_interface/templates/analysis_plugins/general_information.html", line 1, in top-level template code
    {% extends "show_analysis.html" %}
  File "/home/tim/FACT_core/src/web_interface/templates/show_analysis.html", line 16, in top-level template code
    <i class="fas fa-{{ icon }}"></i>
  File "/home/tim/FACT_core/src/web_interface/templates/base.html", line 247, in top-level template code
    {% block body %}{% endblock %}
  File "/home/tim/FACT_core/src/web_interface/templates/show_analysis.html", line 314, in block 'body'
    {% block complete_analysis_block %}
  File "/home/tim/FACT_core/src/web_interface/templates/show_analysis.html", line 317, in block 'complete_analysis_block'
    {% block analysis_result %}
  File "/home/tim/FACT_core/src/web_interface/templates/analysis_plugins/general_information.html", line 41, in block 'analysis_result'
    {% block analysis_result_details %}
  File "<template>", line 47, in block 'analysis_result_details'
  File "/usr/local/lib/python3.8/dist-packages/jinja2/utils.py", line 84, in from_obj
    if hasattr(obj, "jinja_pass_arg"):
jinja2.exceptions.UndefinedError: 'None' has no attribute 'items'
[pid: 19441|app: 0|req: 11/52] 127.0.0.1 () {44 vars in 1240 bytes} [Sun Nov 28 13:55:39 2021] GET /analysis/6fa3cfd7331d28a2f417f0b8b566dd625157d0d8d5f43452c3892195d1a443f1_1579561/elf_analysis/ro/6fa3cfd7331d28a2f417f0b8b566dd625157d0d8d5f43452c3892195d1a443f1_1579561 => generated 290 bytes in 110 msecs (HTTP/1.1 500) 4 headers in 243 bytes (1 switches on core 0)
[pid: 19435|app: 0|req: 12/53] 127.0.0.1 () {40 vars in 887 bytes} [Sun Nov 28 13:55:40 2021] GET /favicon.ico => generated 232 bytes in 1 msecs (HTTP/1.1 404) 4 headers in 231 bytes (1 switches on core 0)
```

Not sure what's going on here

jstucke, 2021-11-29:
Hi, #701 should fix this bug.

jstucke, 2021-11-29:
Please re-open the issue if there are still problems.

PR #701 (2021-11-29): Elf analysis none bugfix
Resolves #700 
- Fixed bug in plugin elf_analysis where `None` is stored in case of no `.modinfo` section
- added check to elf_analysis template so that the server does not crash in case `None` is stored in the DB
- refactoring
closed 2021-11-29 by jstucke