fkie-cad/FACT_core #707: A long pause occurred during the scan, and the Trying loop appeared in the log.  Please take a look.

barkosHuang, 2021-12-01:
1. I uploaded a firmware package. Normal access
2. Then I went to check the system, which was stuck in the current and took a long time.  
[image]
[image]

jstucke, 2021-12-01:
The warnings on the command line ("Trying to remove ... from current analysis ...") is caused by the same file being included multiple  times in one firmware and can safely be ignored. #709 should remove these misleading messages.

It could be the case that some slow analyses are still running and still need some time (e.g. `cwe_checker`). You should be able to see this on the "system health" page under "Analysis Plugins". There are two metrics: the number in the middle (to the right of the cogwheel) shows the analyses that are currently in progress for that plugin and the number right of that shows the queue length. You can already access the other results, though.

barkosHuang, 2021-12-10:
Thank you very much.
I want to pull your code again and update the system again. Is there such a step?

barkosHuang, 2021-12-10:
运行的日志，请问在哪可以找到

barkosHuang, 2021-12-10:
Where can I find the running log

jstucke, 2021-12-10:
> I want to pull your code again and update the system again. Is there such a step?

The easiest and most reliable way to update FACT is to do a `git pull` in the FACT repository folder and then run `src/setup.py` again.

> Where can I find the running log

You can view the log in the command line where you started FACT and also in the web GUI in the navigation bar "Admin" -> "Logs".

barkosHuang, 2021-12-10:
Thank you 
I find that the memory footprint on the back end is very high. Is there any way to optimize it?  
[image]

barkosHuang, 2021-12-10:
The easiest and most reliable way to update FACT is to do a git pull in the FACT repository folder and then run src/setup.py again.
-->
 I can't find the file SRC /setup.py.   src/install.py？

jstucke, 2021-12-10:
Oh sorry for the confusion. Yes, `src/install.py` is the correct file.

jstucke, 2021-12-10:
> I find that the memory footprint on the back end is very high. Is there any way to optimize it?

Since the backend is where all analysis plugins and the unpacking runs, it is a lot more demanding on memory than the other parts. You can try to adjust the worker count for unpacking and the analysis plugins in `src/config/main.cfg`.

```ini
# unpack worker count
[unpack]
threads = 4

# worker count of cpu_architecture plugin
[cpu_architecture]
threads = 2
```

closed 2021-12-13 by barkosHuang