fkie-cad/FACT_core #710: User management

frakman1, 2021-12-01:
The documentation page's screenshot shows a user management feature.
I was unable to find this page in my docker based installation. Is it implemented?

[image]

https://fkie-cad.github.io/FACT_core/_images/10_user_management.png

rhelmke, 2021-12-02:
Hey,

check out [this](https://github.com/fkie-cad/FACT_core/wiki/Authentication) article:

For this to work you must globally enable authentication, which you can do by setting `authentication=true` in `src/config/main.cfg`. Then create your admin user by running the helper CLI in `src/manage_users.py`. Remember to assign the `superuser` role to your created user via the `add_role_to_user` command after creation. Now you can start FACT and will be greeted with a login prompt. Once logged in, you will see the `Manage Users` item under the `Admin` tab.

rhelmke, 2021-12-20:
Hello,

I hope I was able to help you here. Do not hesitate to reopen this ticket if the problem still persists.

closed 2021-12-20 by rhelmke