fkie-cad/FACT_core #712: Make Dependency Graph Searchable

frakman1, 2021-12-03:
The Dependency graph is nice but of limited use for large products. If I can see the module/library/executable, then I can click on it and follow the links. However with hundreds of circles, there's no way of finding the module I am looking for.

[image]

Suggestion is to either allow CTL+F style text searching, or add a search widget that will highlight the circles that match the string being searched.

dorpvom, 2021-12-16:
Cue:

[image]

But for real: We'll look into this, but on first glance it seems to be more complicated than adding a few lines of code. But we might be able to add a search form. As non of us is a JS pro it might be a minute until we come around to do this though.

rhelmke, 2021-12-22:
Hey,

so I am no web-pro, but I made an effort to re-implement the whole dependency graph functionality. You can find the code in pull request #725. It won't be merged before next January as most of us are already in the holidays. But feel free to check it out once it is merged into master.

Cheers

closed 2021-12-22 by rhelmke