fkie-cad/FACT_core #717: How to restart when the project is closed

cen6667, 2021-12-14:
Using ```$ ./start_all_installed_fact_components``` in the docker container will cause errors.
Thank you very much for your reply.

The log shows:
root@fact:/var/log/fact# cd /opt/FACT_core/
root@fact:/opt/FACT_core# ./start_all_installed_fact_components
[2021-12-14 08:49:21][start_all_installed_fact_components][INFO]: starting db
[2021-12-14 08:49:22][MongoMgr][INFO]: Starting local mongo database
[2021-12-14 08:49:22][MongoMgr][INFO]: Starting DB: mongod --auth --config /opt/FACT_core/src/config/mongod.conf --fork --logpath /var/log/fact/mongo.log
Traceback (most recent call last):
  File "/opt/FACT_core/src/../start_fact_db", line 44, in <module>
    FactDb().main()
  File "/opt/FACT_core/src/../start_fact_db", line 36, in __init__
    super().__init__()
  File "/opt/FACT_core/src/fact_base.py", line 31, in __init__
    self.work_load_stat = WorkLoadStatistic(config=self.config, component=self.COMPONENT)
  File "/opt/FACT_core/src/statistic/work_load.py", line 21, in __init__
    self.db = StatisticDbUpdater(config=self.config)
  File "/opt/FACT_core/src/storage/db_interface_statistic.py", line 15, in __init__
    super().__init__(config=config)
  File "/opt/FACT_core/src/storage/mongo_interface.py", line 25, in __init__
    self._authenticate()
  File "/opt/FACT_core/src/storage/mongo_interface.py", line 40, in _authenticate
    self.client.admin.authenticate(user, pw, mechanism='SCRAM-SHA-1')
  File "/usr/local/lib/python3.8/dist-packages/pymongo/collection.py", line 2579, in __call__
    raise TypeError("'Collection' object is not callable. If you "
TypeError: 'Collection' object is not callable. If you meant to call the 'authenticate' method on a 'Database' object it is failing because no such method exists.
[2021-12-14 08:49:23][start_all_installed_fact_components][INFO]: starting frontend
[2021-12-14 08:49:23][start_all_installed_fact_components][INFO]: starting backend
Traceback (most recent call last):
  File "/opt/FACT_core/src/../start_fact_frontend", line 83, in <module>
    FactFrontend().main()
  File "/opt/FACT_core/src/../start_fact_frontend", line 62, in __init__
    super().__init__()
  File "/opt/FACT_core/src/fact_base.py", line 31, in __init__
    self.work_load_stat = WorkLoadStatistic(config=self.config, component=self.COMPONENT)
  File "/opt/FACT_core/src/statistic/work_load.py", line 21, in __init__
    self.db = StatisticDbUpdater(config=self.config)
  File "/opt/FACT_core/src/storage/db_interface_statistic.py", line 15, in __init__
    super().__init__(config=config)
  File "/opt/FACT_core/src/storage/mongo_interface.py", line 25, in __init__
    self._authenticate()
  File "/opt/FACT_core/src/storage/mongo_interface.py", line 40, in _authenticate
    self.client.admin.authenticate(user, pw, mechanism='SCRAM-SHA-1')
  File "/usr/local/lib/python3.8/dist-packages/pymongo/collection.py", line 2579, in __call__
    raise TypeError("'Collection' object is not callable. If you "
TypeError: 'Collection' object is not callable. If you meant to call the 'authenticate' method on a 'Database' object it is failing because no such method exists.
Traceback (most recent call last):
  File "/opt/FACT_core/src/../start_fact_backend", line 91, in <module>
    FactBackend().main()

cen6667, 2021-12-14:
& How to close the project in addition to using ``` Ctrl + c```?
"or by sending a SIGTERM to the start_all_installed_fact_components script."
What exactly is the SIGTERM?How to use？

jstucke, 2021-12-14:
Is this actually a [FACT_docker](https://github.com/fkie-cad/FACT_docker/) issue or did you manually install FACT in a docker container? If you use FACT_docker then you should not have to manually run `start_all_installed_fact_components`. Instead you can simply use the `start.py` script with `./start.py run`.

The error you posted should be related to a new incompatible `pymongo` version which should be fixed with #708,  so if you use the current master branch, it should work.

> & How to close the project in addition to using Ctrl + c?
"or by sending a SIGTERM to the start_all_installed_fact_components script."
What exactly is the SIGTERM?How to use？

If you have problems stopping the docker container, you can simply run `docker stop <container_name>` (you can find out the container name with `docker ps`).
SIGTERM is a kind of [signal](https://en.wikipedia.org/wiki/Signal_(IPC)) which can be sent to a process in order to terminate (stop) it. The signal is intercepted by the start script, though, and a different signal (SIGUSR1) is sent instead in order to cleanly shut down each component. It may be that passing the signal from outside the container (when pressing CTRL+C on from the command line) does not work correctly.

jstucke, 2022-12-16:
Since there is no activity I will close this issue. Please feel free to reopen the issue if the problem persists or if you have additional questions.