fkie-cad/FACT_core #718: key error in clear_recently_finished

wideglide, 2021-12-14:
I had the error below, but I'm not exactly sure what happened.  The `dict` that is being cleared should be multi-processing safe, but somehow between https://github.com/fkie-cad/FACT_core/blob/588f81fa3bd0372567d8b1381bd710735dd5c2fc/src/scheduler/analysis_status.py#L104 and https://github.com/fkie-cad/FACT_core/blob/588f81fa3bd0372567d8b1381bd710735dd5c2fc/src/scheduler/analysis_status.py#L106 the item disappeared.

Is it safe to just add a default value for the `pop` operation?

```python
[2021-12-13 23:32:22][analysis][INFO]: Analysis Completed:
UID: 7092708acc9fb34152f98374602cd722c5177e7b36da306e90320872881f5f07_1344
 Processed analysis: ['unpacker', 'file_type', 'software_components']
 Files included: set()
[2021-12-13 23:32:22][analysis][INFO]: Analysis Completed:
UID: f56f6af2ae0f2c8d7262c01ad3b5c864484e10151ccfb6d417600e7f99d7fae5_7393
 Processed analysis: ['unpacker', 'file_type', 'software_components']
 Files included: set()
Process unpack-load-Worker-None:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/opt/FACT_core/src/helperFunctions/process.py", line 57, in run
    raise exception
  File "/opt/FACT_core/src/helperFunctions/process.py", line 52, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 92, in _work_load_monitor
    workload = self._get_combined_analysis_workload()
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 112, in _get_combined_analysis_workload
    workload = self.get_analysis_workload()
  File "/opt/FACT_core/src/scheduler/analysis.py", line 173, in get_scheduled_workload
    self.status.clear_recently_finished()
  File "/opt/FACT_core/src/scheduler/analysis_status.py", line 106, in clear_recently_finished
    self.recently_finished.pop(uid)
  File "<string>", line 2, in pop
  File "/usr/lib/python3.8/multiprocessing/managers.py", line 850, in _callmethod
    raise convert_to_error(kind, result)
KeyError: '788f9adaa8ddeb0079e39445c0dcc8188e06b3c1c64827630d09ca8926dc0a2e_54526101'
[2021-12-13 23:32:22][process][ERROR]: Exception in unpack-load process:
Traceback (most recent call last):
  File "/opt/FACT_core/src/helperFunctions/process.py", line 52, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 92, in _work_load_monitor
    workload = self._get_combined_analysis_workload()
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 112, in _get_combined_analysis_workload
    workload = self.get_analysis_workload()
  File "/opt/FACT_core/src/scheduler/analysis.py", line 173, in get_scheduled_workload
    self.status.clear_recently_finished()
  File "/opt/FACT_core/src/scheduler/analysis_status.py", line 106, in clear_recently_finished
    self.recently_finished.pop(uid)
  File "<string>", line 2, in pop
  File "/usr/lib/python3.8/multiprocessing/managers.py", line 850, in _callmethod
    raise convert_to_error(kind, result)
KeyError: '788f9adaa8ddeb0079e39445c0dcc8188e06b3c1c64827630d09ca8926dc0a2e_54526101'
```

jstucke, 2021-12-15:
Hmm this is pretty weird. I wonder if this is related to a recent change where this function was extracted into a new module with a couple of others (#713). I will look into it. Ignoring the error with a default values for pop should be fine (since this only clears the list of recently finished analyses for the "system health" page).

jstucke, 2021-12-15:
I was not able to reproduce the error but I noticed that the function is (unintentionally) called by two different processes, so it seems there is a (unlikely) chance of both trying to clear the dictionary at the same time. #719 should fix this, though.

PR #719 (2021-12-15): Current analyses keyerror bugfix
resolves #718 

- added thread safety precautions to `clear_recently_finished()` function
- moved workload computation from unpacking to analysis scheduler
- pylint + typo fixes
closed 2021-12-15 by dorpvom