fkie-cad/FACT_core #737: 500 Internal Server Error

OYoungZF, 2022-02-08:
Hi!

I used the upload function to display the Internal Server Error after the installation was completed using the` /FACT_core/src/install.py`

A terminal prompt:

[2022-02-08 09:07:59,162] ERROR in app: Exception on /upload [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 1518, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.6/dist-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 1516, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 1502, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "./web_interface/components/io_routes.py", line 48, in get_upload
    analysis_plugins = sc.get_available_analysis_plugins()
  File "./intercom/front_end_binding.py", line 37, in get_available_analysis_plugins
    raise Exception('No available plug-ins found. FACT backend might be down!')
Exception: No available plug-ins found. FACT backend might be down!
[2022-02-08 09:07:59][app][ERROR]: Exception on /upload [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 1518, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.6/dist-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 1516, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.6/dist-packages/flask/app.py", line 1502, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "./web_interface/components/io_routes.py", line 48, in get_upload
    analysis_plugins = sc.get_available_analysis_plugins()
  File "./intercom/front_end_binding.py", line 37, in get_available_analysis_plugins
    raise Exception('No available plug-ins found. FACT backend might be down!')
Exception: No available plug-ins found. FACT backend might be down!
[pid: 26464|app: 0|req: 4/19] 127.0.0.1 () {42 vars in 748 bytes} [Tue Feb  8 09:07:59 2022] GET /upload => generated 290 bytes in 32 msecs (HTTP/1.1 500) 2 headers in 99 bytes (1 switches on core 1)

This problem has bothered me for a long time. I am looking forward to your reply!

OYoungZF, 2022-02-08:
And this is my installation error:

`[2022-02-08 09:05:30][install][ERROR]: Failed to run python3 setup_repository.py --update:

Traceback (most recent call last):
  File "src/install.py", line 183, in <module>
    install()
  File "src/install.py", line 151, in install
    install_fact_components(args, distribution, none_chosen, skip_docker)
  File "src/install.py", line 171, in install_fact_components
    backend(skip_docker, distribution)
  File "/home/oyzf/FACT_core/src/install/backend.py", line 45, in main
    _install_plugins(distribution, skip_docker)
  File "/home/oyzf/FACT_core/src/install/backend.py", line 113, in _install_plugins
    plugin_installer.install()
  File "/home/oyzf/FACT_core/src/plugins/installer.py", line 29, in install
    self.install_files()
  File "/home/oyzf/FACT_core/src/plugins/analysis/cve_lookup/install.py", line 29, in install_files
    run_cmd_with_logging('python3 setup_repository.py --update')
  File "/home/oyzf/FACT_core/src/helperFunctions/install.py", line 222, in run_cmd_with_logging
    raise err
  File "/home/oyzf/FACT_core/src/helperFunctions/install.py", line 216, in run_cmd_with_logging
    subprocess.run(cmd_, stdout=PIPE, stderr=PIPE, encoding='UTF-8', shell=shell, check=True, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python3', 'setup_repository.py', '--update']' died with <Signals.SIGKILL: 9>.
`

jstucke, 2022-02-09:
I'm not sure if I understand this correct: Did the installation run completely or did it cancel with the error from your second post? 
The error "Exception: No available plug-ins found. FACT backend might be down!" means that the backend did not store the list of available plugins, so it probably did not start correctly. This could happen if there was an error during startup or if the backend is not installed correctly.

You can safely run the installation script again if it did not complete on the first try. If the installation fails again with this error, you can try to remove the responsible plugin (cve_lookup) from the folder `src/plugins/analysis/` and try again. 

How long did you wait until you interrupted the plugins' installation? It needs to download the CVE database and this may take some time.

OYoungZF, 2022-02-09:
> I'm not sure if I understand this correct: Did the installation run completely or did it cancel with the error from your second post? The error "Exception: No available plug-ins found. FACT backend [...]

Thank you very much for your reply. I re-installed in Ubuntu20.04 and the problem has been fixed. It seems to be version compatibility and network reasons.

closed 2022-02-09 by OYoungZF