fkie-cad/FACT_core #738: Fact Install error - Failed to initialize radare container

scottthomas007, 2022-02-08:
Bonjour,

[2022-02-08 13:16:10][frontend][INFO]: Initializing docker container for radare
Traceback (most recent call last):
  File "src/install.py", line 183, in <module>
    install()
  File "src/install.py", line 151, in install
    install_fact_components(args, distribution, none_chosen, skip_docker)
  File "src/install.py", line 167, in install_fact_components
    frontend(skip_docker, not args.no_radare, args.nginx, distribution)
  File "/home/maverick/FACT_core/src/install/frontend.py", line 189, in main
    _install_docker_images(radare)
  File "/home/maverick/FACT_core/src/install/frontend.py", line 164, in _install_docker_images
    **raise InstallationError('Failed to initialize radare container:\n{}'.format(output))
helperFunctions.install.InstallationError: Failed to initialize radare container:**
app uses an image, skipping
Building server
Sending build context to Docker daemon  4.096kB
Step 1/2 : FROM nginx:alpine
Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
Service 'server' failed to build : Build failed

Regards
Scotty

jstucke, 2022-02-08:
Hi, sounds like you have problems getting a connection to the docker registry. Could this be a proxy problem?

scottthomas007, 2022-02-08:
Thanks for the response. It's not a proxy issue, the docker URL was accessible from the machine. Any other idea?

jstucke, 2022-02-09:
Building the container seems to run for me without errors. Can you try to run `docker-compose build` in the directory `src/install/radare` and report what happens? If it completes without issues, you can simply restart the installation script again.

jstucke, 2022-12-16:
Since there is no activity I will close this issue. Please feel free to reopen the issue if the problem persists or if you have additional questions.

closed 2022-12-16 by jstucke