fkie-cad/FACT_core #747: I may have found the problem about ClamAV in OMS

cen6667, 2022-03-11:
“common_analysis_oms/common_analysis_oms/plugins/ClamAV.json”
There may be an error in the terminal command.

# clamdscan './data/eicar'
[image]

but
# clamscan './data/eicar'
[image]

jstucke, 2022-03-11:
Hi, the message "Could not connect to clamd" usually means that the clamav-daemon service is not running. `clamdscan` uses the clamav daemon service (which should be running in the background) for scanning, which is much faster than `clamscan`. You should be able to check the status with `service clamav-daemon status`. If it is not running, it should be possible to start it with `sudo service clamav-daemon start` (or restart it with `sudo service clamav-daemon restart` if it crashed). 

If this doesn't work, clamav may not be installed correctly. You can try to reinstall it with `python3 src/plugins/analysis/oms/install.py`.

jstucke, 2022-12-16:
Since the plugin was removed in #732 I will close the issue.

closed 2022-12-16 by jstucke