fkie-cad/FACT_core #751: http return "Internal Server Error"

chenjianquan7, 2022-03-25:
FACT run success，but when i restart FACT，it return error ，i cant access http
[image]
[image]

jstucke, 2022-03-25:
The "Plugin has no code directory" messages can be safely ignored (the plugin system tries to load all folders in the plugin folder as plugins but the folder `__pycache__` is an internal python folder and not a plugin which leads to this message). The actual problem that led to the error seems to have happened before the part that you posted. It would be helpful if you could look for a stack trace in the logs.

rhelmke, 2022-09-15:
Issuer didn't respond anymore. Closing.