fkie-cad/FACT_core #752: I got an error  while building docker image of radare

hezygo, 2022-03-25:
here my error,and i use root 
[image]

jstucke, 2022-03-25:
This seems to be an issue with either docker or docker-compose. Could you simply try to navigate to the folder `src/install/radare` and run `docker-compose build --no-cache` and see if it fails? If it does not fail, you could simply try to restart FACT. If it does fail and you don't want to use the radare GUI anyway, you could also comment out the line 
```python
run_cmd_with_logging(f'docker-compose -f {COMPOSE_YAML} up -d')
```
 in `src/start_fact_frontend.py`.

hezygo, 2022-03-25:
> This seems to be an issue with either docker or docker-compose. Could you simply try to navigate to the folder `src/install/radare` and run `docker-compose build --no-cache` and see if it fails? If [...]

i so sad that i try both and got same result

jstucke, 2022-03-25:
Your issue seems to be discussed [here](https://github.com/docker/cli/issues/2971)

Do you need the radare GUI, though? If not simply removing/commenting out the call to docker-compose is still an option (it is just a feature on the FACT analysis page, nothing else depends on it)

jstucke, 2022-12-16:
Since there is no activity I will close this issue. Please feel free to reopen the issue if the problem persists or if you have additional questions.

closed 2022-12-16 by jstucke