fkie-cad/FACT_core #757: Firmware not uploading?

akarag, 2022-03-30:
Hello,
I'm trying to upload a firmware binary (~15mb) but as multiple hours have passed I still get "

jstucke, 2022-03-31:
Hi Alex,
I was not able to reproduce your problem. Did you see any error messages or stack traces in the logs on the command line or in `/tmp/fact_*.log`? 
The upload should definitely not take that long. For a small file (like 1 MB) it should happen almost instantly.

akarag, 2022-04-03:
> 
There were no error messages during install, launch and upload process. I have taken a look into the fact*_logs in the tmp directory and except from the fact_mongo.log they are all empty.

I have copy-pasted my console output into the coonsole.txt and also attached my fact_mongo.log. Their last lines are only repeating. 

[fact_mongo.log](https://github.com/fkie-cad/FACT_core/files/8403997/fact_mongo.log)
[console.txt](https://github.com/fkie-cad/FACT_core/files/8403998/console.txt)

BOerAck, 2022-04-03:
Running into the same issue, Pac-Man animation hangs on upload, terminal reports 0 in queue. I've only tested <10mb files. 

This is my first time installing/using FACT. Running on a fresh VM.

compiled with version: 9.4.0 on 03 April 2022
os: Linux-5.13.0-39-generic 20.04.1-Ubuntu

No tracebacks, closest thing to an error I noticed on start-up:
``
!!! no internal routing support, rebuild with pcre support !!!
``

jstucke, 2022-04-04:
Could you try uploading a file using the REST endpoint? You could use this script I wrote for that:

<details>

```python
from argparse import ArgumentParser
from base64 import b64encode
from pathlib import Path

import requests

def parse_args():
    parser = ArgumentParser()
    parser.add_argument("file", type=Path, help="the file to upload")
    parser.add_argument("device_name", type=str, help="the kind of device")
    parser.add_argument("device_class", type=str, help="the type of device (e.g. router)")
    parser.add_argument("vendor", type=str, help="the vendor / manufacturer of the device")
    parser.add_argument("version", type=str, help="the version of the firmware")
    parser.add_argument("release_date", type=str, help="the release date of the firmware (YYYY-MM-DD)")
    parser.add_argument("--device_part", type=str, help="part of the firmware (complete, kernel, bootloader or root-fs)",
                        default="complete")
    parser.add_argument("-u", "--url", type=str, default="http://localhost:5000/",
                        help="the URL of your FACT instance (e.g. http://localhost:5000/)")
    parser.add_argument("-p", "--plugins", type=str, default="",
                        help="a comma-separated list with the selected analysis plugins "
                             "(e.g. \"device_tree,kernel_config\") (optional)")
    parser.add_argument("-t", "--tags", type=str, default="", help="a comma-separated list of tags (optional)")
    return parser.parse_args()

def main(options):
    assert options.file.is_file(), f"Error: file not found {options.file}"

    response = requests.put(options.url.rstrip("/") + "/rest/firmware", json={
        "device_name": options.device_name,
        "device_part": options.device_part,
        "device_class": options.device_class,
        "file_name": options.file.name,
        "version": options.version,
        "vendor": options.vendor,
        "release_date": options.release_date,
        "tags": options.tags if options.tags else "",
        "requested_analysis_systems": options.plugins.split(",") if options.plugins else [],
        "binary": b64encode(options.file.read_bytes()).decode()
    })

    print(response.json())

if __name__ == '__main__':
    main(parse_args())

```

</details>

akarag, 2022-04-04:
> 

Thank you very much, your uploadscript works. I was able to upload the firmware into FACT and I can also see it under Latest Firmware Submissions and perform further analysis.

akarag, 2022-04-04:
I have tried it just now, in fact nothing happens. I get to the Pacman animation, but the Network tab remains empty.

jstucke, 2022-04-05:
I tried to fix this in #760 but since neither I nor @rhelmke can reproduce the problem, it would be great if you could confirm whether or not this fixes the problem. All you need to do is check out the branch (in the FACT folder), restart FACT and try to upload something: `git checkout upload-submit-bugfix`

akarag, 2022-04-05:
Thanks for your bugfix, now an upload via the web interface also works and I can also see the POST request in the console:

[image]

jstucke, 2022-04-06:
That is great to hear. Thank you for helping with debugging!

rhelmke, 2022-04-06:
Thank you @akarag! We just merged the fix into master :-)

PR #760 (2022-04-06): fix upload submit bug
Fixed a bug in the upload template where the POST request to `/upload` was not triggered on some systems
Resolves #750, resolves #757
closed 2022-04-06 by rhelmke