fkie-cad/FACT_core #765: I got an error when i try to run a `software_components` plugin

hezygo, 2022-04-08:
here my main code
[image]
and here my log
[image]
is  the error caused by my input args?

rhelmke, 2022-04-08:
Hello @hezygo.

It seems like you are trying to extract selected code components of FACT and interface with them using your own standalone script. While that is totally fine, we kindly remind you that FACT is a tightly integrated application that depends on many different components. It is very likely that something will break when you interface with single components. 

We understand that FACT is far from complete in its use case and, thus, provide well-defined APIs for interaction with other tools to mitigate possible problems. For these, we gladly provide support and are open for new suggestions and pull requests. However, what you are trying to achieve is beyond the scope of our project.

Thus, I'm closing this issue.

closed 2022-04-08 by rhelmke

jstucke, 2022-04-08:
Some pointers:
- I'm not really sure what you are trying to accomplish but the plugins are not intended to be used outside of FACT
- all the yara rules of the software_components plugin are compiled into `src/analysis/signatures/software_components.yc` by the script `src/compile_yara_signatures.py`
- The rule "no_text_file" uses yara's magic module so you need to compile yara with magic support (see https://yara.readthedocs.io/en/v4.2.0/gettingstarted.html#compiling-yara)