fkie-cad/FACT_core #794: Problem with Docker-Containers when running in User Remap Mode.

SonicFM, 2022-06-11:
Hi,

i get a error when i try to analyse an image everytime a container is involved, such as the unpacker-Container.

```
[2022-06-10 17:51:16][docker][WARNING]: [Docker]: encountered docker error while processing
[2022-06-10 17:51:20][process][ERROR]: ESC[91mException in Unpacking process:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/docker/api/client.py", line 268, in _raise_for_status
    response.raise_for_status()
  File "/usr/local/lib/python3.8/dist-packages/requests/models.py", line 960, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: Bad Request for url: http+docker://localhost/v1.41/containers/create

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/FACT_core/src/helperFunctions/process.py", line 52, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/FACT_core/src/scheduler/Unpacking.py", line 70, in unpack_worker
    extracted_objects = unpacker.unpack(fo)
  File "/opt/FACT_core/src/unpacker/unpack.py", line 39, in unpack
    extracted_files = self.extract_files_from_file(file_path, tmp_dir.name)
  File "/opt/FACT_core/src/unpacker/unpack_base.py", line 26, in extract_files_from_file
    result = run_docker_container(
  File "/opt/FACT_core/src/helperFunctions/docker.py", line 35, in run_docker_container
    container = client.containers.run(image, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/docker/models/containers.py", line 819, in run
    container = self.create(image=image, command=command,
  File "/usr/local/lib/python3.8/dist-packages/docker/models/containers.py", line 878, in create
    resp = self.client.api.create_container(**create_kwargs)
  File "/usr/local/lib/python3.8/dist-packages/docker/api/container.py", line 428, in create_container
    return self.create_container_from_config(config, name)
  File "/usr/local/lib/python3.8/dist-packages/docker/api/container.py", line 439, in create_container_from_config
    return self._result(res, True)
  File "/usr/local/lib/python3.8/dist-packages/docker/api/client.py", line 274, in _result
    self._raise_for_status(response)
  File "/usr/local/lib/python3.8/dist-packages/docker/api/client.py", line 270, in _raise_for_status
    raise create_api_error_from_http_exception(e)
  File "/usr/local/lib/python3.8/dist-packages/docker/errors.py", line 31, in create_api_error_from_http_exception
    raise cls(e, response=response, explanation=explanation)
docker.errors.APIError: 400 Client Error for http+docker://localhost/v1.41/containers/create: Bad Request ("privileged mode is incompatible with user namespaces.  You must run the container in the host namespace when running privileged mode")
ESC[0m
[2022-06-10 17:51:20][process][WARNING]: ESC[93mrestarting Unpacking 5 processESC[0m

```
Docker Deamon Config:
```
{
    "userns-remap": "default",
    "live-restore": true,
    "userland-proxy": false,
    "no-new-privileges": true,
    "ipv6": true,
    "fixed-cidr-v6":"fd00:dead:beef:c0::/80",
    "experimental": true,
    "ip6tables": true
}
```
I do this to improve security, since im running other Containers as well. Since my other Containers are using a Docker-Compose yaml i can specify "userns_mode: host" to enable the privileged mode.

But the FACT-Containers are spawned with "docker run"?! Therefore i asume i cannot fix this by myself.

If there is an option to get this working i would appreciate it!

Thanks for yours awesome work on FACT!

jstucke, 2022-06-14:
The extractor container needs "privileged mode" to be able to mount file systems (to extract files from those). Maybe there is some way to accomplish this without relying on privileged mode (maybe something like `guestmount` or `udisksctl`). 
I fear I cannot really help you with "User Namespace Remapping" because I have not used it before. Do we even need privileged mode if namespace remapping is activated?

SonicFM, 2022-07-24:
Thanks for the reply and sorry for my late response.
"Do we even need privileged mode if namespace remapping is activated?"
Sadly i cannot answer this. But i can tell for my experience with other Containers using the "userns_mode:host" disables the namespace remapping thus enabling privileged mode, since the container is running as root, if i understood this correctly.

Maybe adding "--userns=host" to the "docker run" command could solve this issue?! Can you tell me the exact "docker run" command which is executed and how i could modify it? Then i could test this on my Machine.
Ref.: https://docs.docker.com/engine/security/userns-remap/ "Disable namespace remapping for a container"

jstucke, 2022-08-01:
You can find the call to the extractor container in `src/unpacker/unpack_base.py:26` but it uses the Docker Python API. The corresponding option here seems to be `userns_mode`. Replacing `privileged=True` with `userns_mode='host'` should work. 

I tried it but mounting seems to fail:
> mount: /tmp/tmp2twdonlf: failed to setup loop device for /tmp/extractor/input/11863f39512c710d0a640d06582a08858e5e0c557a8327a6eb1d07d2fd3baf82 10485760.
cp: cannot stat '/tmp/tmp2twdonlf/*': No such file or directory
umount: /tmp/tmp2twdonlf: must be superuser to unmount.

SonicFM, 2022-08-01:
Thanks for your feedback!

as stated in https://github.com/fkie-cad/FACT_docker/issues/23 i'll try to setup Namespace-Remapping on my non-dockerized FACT. Maybe i'll get any new information.

jstucke, 2022-08-02:
So the way I understand it now, we could set `userns_mode='host'` (to disable user namespace remapping) in combination with `privileged=True` to fix the issues you are experiencing with the extractor container and this would have no effects for other users (who don't have namespace remapping enabled in the first place), right? Then this would probably be a good solution as long as we don't find a way to make privileged mode obsolete for the extractor.

maringuu, 2022-09-06:
I did investigate a bit.
As @jstucke suggested replacing `privileged=True` by `userns_mode=host` in https://github.com/fkie-cad/FACT_core/blob/8fdee848cdf3e02d4c98458a4a40077b335e2262/src/unpacker/unpack_base.py#L26-L36 works around the error that privileged mode cannot be used when namespace remapping is enabled.
My cli looked like this:
```sh
dkr run \                                                                                                                       1 ↵
    --rm \
    --userns=host \
    --mount type=bind,source=/dev,destination=/dev \
    --mount type=bind,source=$HOME/tmp/extract,destination=/tmp/extractor \
    --memory=1024m \
    fkiecad/fact_extractor --chown 1000:1000
```

What causes the error is the `--chown 1000:1000`. With remapping enabled the container simply can't chown to a user outside of the subuid range.