fkie-cad/FACT_core #795: Logfiles get other names than specified

SonicFM, 2022-06-11:
Hi,

i have configured systemd-services for FACT in order to control the state of the DB, Frontend and Backend better.
I specified the wanted logfile names and the directory.

So far this is working, but if i look into the Logs-folder i notice these files get renamed.

For example:
```
[Unit]
Description=FACT - Frontend
After=network.target

[Service]
Type=simple
Restart=always
User=fact
Group=fact
ExecStart=/opt/FACT_core/src/start_fact_frontend.py -l /opt/FACT_core/logs/fact_frontend.log -L WARNING -s

[Install]
WantedBy=multi-user.target
```

/opt/FACT_core/logs/fact_frontend.log gets somehow converted into "fact_frontend_frontend.log" and logs are written into this file.

The same happens for Backend and DB:
fact_db.log -> fact_db_database.log but strangely logs are written to "fact_db.log" this time...
fact_backend.log -> fact_backend_backend.log

Also i can see that the DB-Logs gets somehow rotated, but not via logrotate.

```
fact_db.log
fact_db.log.2022-06-10T15-00-04
fact_db.log.2022-06-10T15-35-43
fact_db.log.2022-06-11T06-00-02
...
```
What am i doing wrong? Or do i missed a cruzial setting?

Thanks for your work on FACT!

jstucke, 2022-06-14:
At some point we added individual logs for cases when FACT is not set up in a distributed way. It works by using the component name ("frontend", "backend" or "db") and appending it to the name of the configured log file (it happens in `get_log_file_for_component()` in `src/helperFunctions/program_setup.py`). So in theory it should work if you pass only something like `fact.log` as log file name. 
I'm not sure what causes the log rotation. At least it doesn't seem to be the python logging module.

SonicFM, 2022-07-24:
Thanks for the reply and sorry for my late reply!
I have tested your suggestions and can report the following:

In the SystemD service file i have changed
"ExecStart=/opt/FACT_core/src/start_fact_frontend.py -l /opt/FACT_core/logs/fact_[frontend/backend/db].log -L WARNING -s"
to
"ExecStart=/opt/FACT_core/src/start_fact_frontend.py -l /opt/FACT_core/logs/fact.log -L WARNING -s"

systemctl daemon-reload
and started the corresponding services.

The following logs then showed up in the specified folder:

fact_backend.log
fact_database.log (gets created but is never written to)
fact_db.log (gets created and is written to)
fact_frontend.log
fact.log (gets created but is never written to)

If applicable, the "-l" option to specify the log might then not be path specification + log name but exclusively a path specification if the logs get their own name anyway.
e.g.: ExecStart=/opt/FACT_core/src/start_fact_frontend.py **-l /opt/FACT_core/logs/** -L WARNING -s

It may be worth documenting this behavior if it hasn't already been done and if I haven't missed it.

jstucke, 2022-08-01:
Sorry for the late response (I was out of office the last week). 

> If applicable, the "-l" option to specify the log might then not be path specification + log name but exclusively a path specification if the logs get their own name anyway.

I agree that it is not very intuitive the way it is now and we will probably fix this in the near future. As far as I know, there is no documentation for this (apart from the `--help` option of the CLI scripts)

SonicFM, 2022-08-01:
You do not need to apologize :)

Thank you very much for your feedback. I'm looking forward to it until this issue is fixed. It's not a game changer, but it may confuse others aswell :)

jstucke, 2024-11-13:
fixed in #850

commit c1f0d255 (2022-09-08): logging: Respect file name chosen by user

Previously {frontend,backend,database} were appended for the respective components.
Now the name as chosen by the -l argument is used.

When -l is not specified the name as given in the config is used.

Also this commit removes `helperFunctions.program_setup.program_setup` as it
did not do anything besides call `setup_argparser` and `setup_logging`.
The signatures of these functions were changed to allow for more flexibility.

Closes #795
closed 2024-11-13 by jstucke