fkie-cad/FACT_core #847: The cve vulnerability in the list does not match the actual version

KongLynn, 2022-08-25:
Hi, I have scanned a firmware using cve lookup plugin, but it seems that the cve vulnerability in the list does not match the actual version, for examaple, my openssl version is 1.0.2u, why CVE-1999-0428 is in the list？It only affects version below 0.9.2b.
[image]

KongLynn, 2022-09-08:
Hi, 
I also find that openssl scans out multiple versions, some versions are not used.what is the scanning principle? The firmware can be download in https://www.tp-link.com/us/support/download/archer-ax11000/#Firmware.
Hope to get a response from you as soon as possible, thank you very much!

jstucke, 2022-09-08:
Hi, some of the matches could be false positives: the YARA signature for OpenSSL is not very rebust and could match in other binaries. That is something that could definately be improved. What is more, OpenSSL / libssl is often linked as library into other binaries during compilation. This could also generate some matches. You can find the signatures in the "software_components" plugin. You could try to investigate which individual files matched the signature and try to find the "real" match.

jstucke, 2024-11-13:
Is this still an issue? I tried to reproduce the issue with the exact version of OpenSSL, but did not get any unexpected results.

jstucke, 2024-12-04:
I will close this issue, since it has been stale for quite a while. If this problem persists, feel free to reopen it.

closed 2024-12-04 by jstucke