fkie-cad/FACT_core #849: is analysis file size Limit? why big file has not analysis

chenjianquan7, 2022-09-07:


jstucke, 2022-09-07:
Hi, how large is the file you are trying to analyze? If files are very large, some analysis plugins may run into timeouts, which results in no analysis data being stored in the database and consequently no results appearing on the analysis page.

chenjianquan7, 2022-09-07:
> analyze

for example 3m

jstucke, 2022-09-07:
Do you mean 3 MB? That can still be relatively large depending on the analysis / plugin. E.g. that would be very large for a ELF executable and the CWE-Checker plugin could very well run into a timeout. If a timeout happened, there should be error messages appearing in your logs (e.g. "... Timeout during analysis..."). Did you observe any such messages?

emorasoul, 2022-09-23:
have some crash maybe