fkie-cad/FACT_core #853: Run the update_statistic program.

harry1080, 2022-09-12:
How to update this data

Run the update_statistic program.
[image]

rhelmke, 2022-09-12:
Hey buddy,

please run the mentioned python script in your local FACT folder under [src/update_statistic.py](https://github.com/fkie-cad/FACT_core/blob/master/src/update_statistic.py) while FACT is running.

harry1080, 2022-09-12:
hello buddy,
I have executed this program as you said, but the data is still not displayed data. Why? 

[image]

harry1080, 2022-09-13:
Hello, brother

After reading the articles written by other leaders, I found that there should be a route in this position, as shown in the following figure:

[image]

But the system installed on my side does not have this option. Is this not the default, or is there a problem with my installation,

[image]

rhelmke, 2022-09-13:
Just must analyze firmware before getting statistics about them.

The meta data shown in the first image is not the default. You are supposed to organize your firmware samples by yourself -- meaning you need to select `new entry` and type in your device class.

harry1080, 2022-09-13:
Hey buddy,
Received, thank you for your advice, solved this problem, you can close this issues

rhelmke, 2022-09-13:
Cheers!

closed 2022-09-13 by rhelmke