fkie-cad/FACT_core #856: install with -N option is not working as expected

vijayeswari, 2022-09-20:
Hi, I tried to use the -N option to enable the nginx . Installation is complete, but the curl is **NOT** responding on http://localhost. Is there anything else that needs to be done?

`P2713770@ip-47-228-8-80:~$ curl -v http://localhost/rest/firmware
*   Trying 127.0.0.1:80...
* connect to 127.0.0.1 port 80 failed: Connection refused
*   Trying ::1:80...
* connect to ::1 port 80 failed: Connection refused
* Failed to connect to localhost port 80 after 0 ms: Connection refused
* Closing connection 0
curl: (7) Failed to connect to localhost port 80 after 0 ms: Connection refused`

It only works with localhost:5000 
`P2713770@ip-47-228-8-80:~$ curl -v http://localhost:5000/rest/firmware/c4d3851ec77f3f7661046d8457add8097851cb7d4f5c53bdff2e0ebca80b56d9_29697479
*   Trying 127.0.0.1:5000...
* Connected to localhost (127.0.0.1) port 5000 (#0)
> GET /rest/firmware/c4d3851ec77f3f7661046d8457add8097851cb7d4f5c53bdff2e0ebca80b56d9_29697479 HTTP/1.1
> Host: localhost:5000
> User-Agent: curl/7.81.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: application/json
< Content-Length: 6319
`

vijayeswari, 2022-09-20:
I have FACT 4.0 version installed. The documentation needs to be updated as this version is not using Mongo db. For a new person installing FACT 4.0 for the first time , may be confused as its PostgresSQL , but the documentation is all about mongo db. 

https://github.com/fkie-cad/FACT_core/wiki/radare-integration#ssl-and-hostname-issues

I am following this for nginx (case 1) and exposing the service to network (case 2 ) , but the src/helperFunctions/web_interface.py::get_radare_endpoint  is missing in the web_interface file. Would appreciate if the document is updated for FACT 4.0 code.

dorpvom, 2022-09-20:
Hi vijayeswari,
the nginx reverse proxy should actually be running on https / port 443. Can you check that?

dorpvom, 2022-09-20:
We'll check on the documentation. This should obviously have been migrated in the meantime.

vijayeswari, 2022-09-20:
P2713770@ip-47-228-8-80:~$ curl -v https://localhost:443/rest/firmware
*   Trying 127.0.0.1:443...
* Connected to localhost (127.0.0.1) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (OUT), TLS header, Unknown (21):
* TLSv1.2 (OUT), TLS alert, unknown CA (560):
* SSL certificate problem: self-signed certificate
* Closing connection 0
curl: (60) SSL certificate problem: self-signed certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.

P2713770@ip-47-228-8-80:~$ curl -v http://localhost:5000/rest/firmware
*   Trying 127.0.0.1:5000...
* Connected to localhost (127.0.0.1) port 5000 (#0)
> GET /rest/firmware HTTP/1.1
> Host: localhost:5000
> User-Agent: curl/7.81.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: application/json
< Content-Length: 330
< 
* Connection #0 to host localhost left intact
{"request": {"inverted": false, "limit": 0, "offset": 0, "query": {}, "recursive": false}, "request_resource": "/rest/firmware", "status": 0, "timestamp": 1663703791, "uids": ["9c7b2b7061f193b24fb79e620d214861cabb3dacd195c7bb88842087c1dc38d7_34675268", "c4d3851ec77f3f7661046d8457add8097851cb7d4f5c53bdff2e0ebca80b56d9_29697479"]}P2713770@ip-47-228-8-80:~$

vijayeswari, 2022-09-20:
Thank you @dorpvom  for your quick reply. I tried https over 443 , it didnt work either. 
`P2713770@ip-47-228-8-80:~$ curl -v https://localhost/rest/firmware
*   Trying 127.0.0.1:443...
* Connected to localhost (127.0.0.1) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (OUT), TLS header, Unknown (21):
* TLSv1.2 (OUT), TLS alert, unknown CA (560):
* SSL certificate problem: self-signed certificate
* Closing connection 0
curl: (60) SSL certificate problem: self-signed certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
P2713770@ip-47-228-8-80:~$ 
`

vijayeswari, 2022-09-20:
I see te self-signed certs are generated and available under /etc/nginx/

`
def _generate_and_install_certificate():
    logging.info('Generating self-signed certificate')
    execute_commands_and_raise_on_return_code([
        'openssl genrsa -out fact.key 4096',
        f'echo "{DEFAULT_CERT}" | openssl req -new -key fact.key -out fact.csr',
        'openssl x509 -req -days 730 -in fact.csr -signkey fact.key -out fact.crt',
        'sudo mv fact.key fact.csr fact.crt /etc/nginx'
    ], error='generate SSL certificate')

`

P2713770@ip-47-228-8-80:/etc/nginx$ ls
conf.d  fact.crt  fact.key      fastcgi_params  koi-win     modules-available  nginx.conf      proxy_params  sites-available  snippets      win-utf
error   fact.csr  fastcgi.conf  koi-utf         mime.types  modules-enabled    nginx.conf.bak  scgi_params   sites-enabled    uwsgi_params

dorpvom, 2022-09-21:
Exactly.
So we generate a self-signed certificate for your convenience. You can either
- keep it and add it to your trust list,
- replace it by one generated yourself (same, trustlist) or
- generate a trusted certificate from a CA.

Though you probably only need the latter if you want to host your FACT on the internet. Our company has an own intermediate CA that is trusted by all company PCs that signs such certificates for internal purposes.

Alternatively you can also modify the nginx config to host the tool on 80.

jstucke, 2024-11-13:
I will close this issue, since it has been stale for quite some time. Feel free to reopen it if the problem is not resolved.

closed 2024-11-13 by jstucke