fkie-cad/FACT_core #857: Internal Server Error when visiting http://127.0.0.1:5000/

1oftheguthakrs, 2022-09-21:
sudo ./start_all_installed_fact_components

[sudo] password for kali: 
[2022-09-21 11:12:30][start_all_installed_fact_components][INFO]: starting db
[2022-09-21 11:12:31][fact_base][INFO]: Successfully started FACT DB-Service
[2022-09-21 11:12:31][fact_base][INFO]: Successfully started FACT DB-Service
[2022-09-21 11:12:32][start_all_installed_fact_components][INFO]: starting frontend
[2022-09-21 11:12:32][start_all_installed_fact_components][INFO]: starting backend
/usr/lib/python3/dist-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
[2022-09-21 11:12:37][fact_base][INFO]: Successfully started FACT Frontend
[uWSGI] getting INI configuration from /opt/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.20 (64bit) on [Wed Sep 21 11:12:37 2022] ***
compiled with version: 12.2.0 on 16 September 2022 15:53:29
os: Linux-5.18.0-kali7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.16-1kali1 (2022-08-31)
nodename: kali
machine: x86_64
clock source: unix
detected number of CPU cores: 4
current working directory: /opt/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
your processes number limit is 63694
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
probably another instance of uWSGI is running on the same address (127.0.0.1:5000).
bind(): Address already in use [core/socket.c line 769]
[2022-09-21 11:12:37][cwe_checker][INFO]: Version is cwe_checker 0.6.0

[2022-09-21 11:12:38][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2022-09-21 11:12:38][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
[2022-09-21 11:12:39][analysis][INFO]: Analysis System online...
[2022-09-21 11:12:39][analysis][INFO]: Plugins available: ['binwalk', 'cpu_architecture', 'crypto_hints', 'crypto_material', 'cve_lookup', 'cwe_checker', 'device_tree', 'dummy_plugin_for_testing_only', 'elf_analysis', 'exploit_mitigations', 'file_hashes', 'file_system_metadata', 'file_type', 'hardware_analysis', 'hashlookup', 'information_leaks', 'init_systems', 'input_vectors', 'interesting_uris', 'ip_and_uri_finder', 'kernel_config', 'known_vulnerabilities', 'malware_scanner', 'printable_strings', 'qemu_exec', 'software_components', 'source_code_analysis', 'string_evaluator', 'tlsh', 'users_and_passwords']
[2022-09-21 11:12:39][unpacking_scheduler][INFO]: Unpacker Module online
[2022-09-21 11:12:39][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2022-09-21 11:12:39][compare][INFO]: Plug-ins available: dict_keys(['File_Coverage', 'File_Header', 'Software'])
[2022-09-21 11:12:39][comparison_scheduler][INFO]: Comparison Scheduler online...
[2022-09-21 11:12:39][binary_service][INFO]: binary service online
[2022-09-21 11:12:39][back_end_binding][INFO]: InterCom started
[2022-09-21 11:12:39][binary_service][INFO]: binary service online
[2022-09-21 11:12:40][binary_service][INFO]: binary service online
[2022-09-21 11:13:31][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0

[image]

dorpvom, 2022-09-21:
Hi 1oftheguthakrs,
It seems that port 5000 is already in use on your system:
```
probably another instance of uWSGI is running on the same address (127.0.0.1:5000).
bind(): Address already in use [core/socket.c line 769]
```
You can use `netstat -tlpen` or sth. similar to check what runs there. You can also configure FACT to use a different port in the `src/config/uwsgi.ini` file.
Another common issue is that FACT was running before and wasn't shut down correctly so that the uwsgi is still running in the background.

1oftheguthakrs, 2022-09-21:
thank you for pointing it out. Indeed, it was another uWSGI.

dorpvom, 2022-09-21:
I'll close this then. Feel free to re-open if additional advice is needed.
Thanks

closed 2022-09-21 by dorpvom