fkie-cad/FACT_core #858: ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security'

1oftheguthakrs, 2022-09-21:
Starting with..

```sudo ./start_all_installed_fact_components```

I get internal service error when opening up 127.0.0.1:5000

[2022-09-21 15:36:51][start_all_installed_fact_components][INFO]: starting db
[2022-09-21 15:36:52][fact_base][INFO]: Successfully started FACT DB-Service
[2022-09-21 15:36:52][fact_base][INFO]: Successfully started FACT DB-Service
[2022-09-21 15:36:53][start_all_installed_fact_components][INFO]: starting frontend
[2022-09-21 15:36:53][start_all_installed_fact_components][INFO]: starting backend
/usr/lib/python3/dist-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
[2022-09-21 15:36:58][fact_base][INFO]: Successfully started FACT Frontend
[uWSGI] getting INI configuration from /opt/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.20 (64bit) on [Wed Sep 21 15:36:58 2022] ***
compiled with version: 12.2.0 on 16 September 2022 15:53:29
os: Linux-5.18.0-kali7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.16-1kali1 (2022-08-31)
nodename: kali
machine: x86_64
clock source: unix
detected number of CPU cores: 4
current working directory: /opt/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
your processes number limit is 63694
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address 127.0.0.1:5001 fd 3
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
Python version: 3.10.7 (main, Sep  8 2022, 14:34:29) [GCC 12.2.0]
Python main interpreter initialized at 0x555f336efb40
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500256 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
[2022-09-21 15:36:58][cwe_checker][INFO]: Version is cwe_checker 0.6.0

[2022-09-21 15:36:59][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2022-09-21 15:36:59][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 27, in <module>
    from web_interface.frontend_main import WebFrontEnd
  File "/opt/FACT_core/src/./web_interface/frontend_main.py", line 6, in <module>
    from web_interface.app import create_app
  File "/opt/FACT_core/src/./web_interface/app.py", line 4, in <module>
    from flask_security import uia_username_mapper
  File "/usr/local/lib/python3.10/dist-packages/flask_security/__init__.py", line 14, in <module>
    from .changeable import admin_change_password
  File "/usr/local/lib/python3.10/dist-packages/flask_security/changeable.py", line 19, in <module>
    from .utils import config_value, hash_password, login_user, send_mail
  File "/usr/local/lib/python3.10/dist-packages/flask_security/utils.py", line 42, in <module>
    from flask_wtf import csrf
  File "/usr/lib/python3/dist-packages/flask_wtf/__init__.py", line 3, in <module>
    from .csrf import CSRFProtect, CsrfProtect
  File "/usr/lib/python3/dist-packages/flask_wtf/csrf.py", line 10, in <module>
    from werkzeug.security import safe_str_cmp
ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security' (/usr/local/lib/python3.10/dist-packages/werkzeug/security.py)
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 101948)
spawned uWSGI worker 1 (pid: 102112, cores: 2)
spawned uWSGI worker 2 (pid: 102113, cores: 2)
spawned uWSGI worker 3 (pid: 102115, cores: 2)
spawned uWSGI worker 4 (pid: 102117, cores: 2)
spawned uWSGI worker 5 (pid: 102118, cores: 2)
*** Stats server enabled on 127.0.0.1:9191 fd: 17 ***
[2022-09-21 15:37:00][analysis][INFO]: Analysis System online...
[2022-09-21 15:37:00][analysis][INFO]: Plugins available: ['binwalk', 'cpu_architecture', 'crypto_hints', 'crypto_material', 'cve_lookup', 'cwe_checker', 'device_tree', 'dummy_plugin_for_testing_only', 'elf_analysis', 'exploit_mitigations', 'file_hashes', 'file_system_metadata', 'file_type', 'hardware_analysis', 'hashlookup', 'information_leaks', 'init_systems', 'input_vectors', 'interesting_uris', 'ip_and_uri_finder', 'kernel_config', 'known_vulnerabilities', 'malware_scanner', 'printable_strings', 'qemu_exec', 'software_components', 'source_code_analysis', 'string_evaluator', 'tlsh', 'users_and_passwords']
[2022-09-21 15:37:00][unpacking_scheduler][INFO]: Unpacker Module online
[2022-09-21 15:37:00][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2022-09-21 15:37:01][compare][INFO]: Plug-ins available: dict_keys(['File_Coverage', 'File_Header', 'Software'])
[2022-09-21 15:37:01][comparison_scheduler][INFO]: Comparison Scheduler online...
[2022-09-21 15:37:01][binary_service][INFO]: binary service online
[2022-09-21 15:37:01][binary_service][INFO]: binary service online
[2022-09-21 15:37:01][back_end_binding][INFO]: InterCom started
[2022-09-21 15:37:01][binary_service][INFO]: binary service online
[2022-09-21 15:37:52][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2022-09-21 15:38:45][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2022-09-21 15:39:37][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
--- no python application found, check your startup logs for errors ---
[pid: 102112|app: -1|req: -1/1] 127.0.0.1 () {40 vars in 644 bytes} [Wed Sep 21 15:40:07 2022] GET / => generated 21 bytes in 0 msecs (HTTP/1.1 500) 2 headers in 83 bytes (0 switches on core 0)

I think it has to do with Werkzeug. I downgraded to 2.1.2 and got a different error with Werkzeug. 

```pip list | grep flask*       ```
flasgger                          0.9.5
flask-paginate                    2022.1.8
flask-restx                       0.5.1

```pip list | grep Werkzeug*```
Werkzeug                          2.1.2

Am I on the right track? And what's would be the right version combi to use?

dorpvom, 2022-09-23:
Hi 1oftheguthakrs,
we frequently encounter errors from werkzeug <-> flask <-> flasx_<extension_x> version conflicts which we try to detect early through our CI. Are you running FACT on current master? If so, we've to do some sleuthing into where this new issue is coming from.

jstucke, 2022-09-26:
Hi, there were problems with flask-restx and the newest version of Werkzeug (see https://github.com/python-restx/flask-restx/pull/463) and it should be fixed now but there still is no new release (meaning that you need to use Werkzeug <=2.1.2 up until then).

> I downgraded to 2.1.2 and got a different error with Werkzeug.

What was the error? Did you also downgrade Flask?

1oftheguthakrs, 2022-09-26:
I am not able to reproduce that error anymore. Now I am getting the same error.

```/start_all_installed_fact_components```
[2022-09-26 10:38:53][start_all_installed_fact_components][INFO]: starting db
[2022-09-26 10:38:54][fact_base][INFO]: Successfully started FACT DB-Service
[2022-09-26 10:38:54][fact_base][INFO]: Successfully started FACT DB-Service
[2022-09-26 10:38:55][start_all_installed_fact_components][INFO]: starting frontend
[2022-09-26 10:38:55][start_all_installed_fact_components][INFO]: starting backend
/usr/lib/python3/dist-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
[2022-09-26 10:39:02][fact_base][INFO]: Successfully started FACT Frontend
[uWSGI] getting INI configuration from /opt/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.20 (64bit) on [Mon Sep 26 10:39:02 2022] ***
compiled with version: 12.2.0 on 16 September 2022 15:53:29
os: Linux-5.18.0-kali7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.16-1kali1 (2022-08-31)
nodename: kali
machine: x86_64
clock source: unix
detected number of CPU cores: 4
current working directory: /opt/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
your processes number limit is 63694
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address 127.0.0.1:5001 fd 3
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
Python version: 3.10.7 (main, Sep  8 2022, 14:34:29) [GCC 12.2.0]
Python main interpreter initialized at 0x55ab2e654f50
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500256 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
[2022-09-26 10:39:03][cwe_checker][INFO]: Version is cwe_checker 0.6.0

[2022-09-26 10:39:04][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2022-09-26 10:39:04][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
[2022-09-26 10:39:06][analysis][INFO]: Analysis System online...
[2022-09-26 10:39:06][analysis][INFO]: Plugins available: ['binwalk', 'cpu_architecture', 'crypto_hints', 'crypto_material', 'cve_lookup', 'cwe_checker', 'device_tree', 'dummy_plugin_for_testing_only', 'elf_analysis', 'exploit_mitigations', 'file_hashes', 'file_system_metadata', 'file_type', 'hardware_analysis', 'hashlookup', 'information_leaks', 'init_systems', 'input_vectors', 'interesting_uris', 'ip_and_uri_finder', 'kernel_config', 'known_vulnerabilities', 'malware_scanner', 'printable_strings', 'qemu_exec', 'software_components', 'source_code_analysis', 'string_evaluator', 'tlsh', 'users_and_passwords']
[2022-09-26 10:39:06][unpacking_scheduler][INFO]: Unpacker Module online
[2022-09-26 10:39:06][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 27, in <module>
    from web_interface.frontend_main import WebFrontEnd
  File "/opt/FACT_core/src/./web_interface/frontend_main.py", line 6, in <module>
    from web_interface.app import create_app
  File "/opt/FACT_core/src/./web_interface/app.py", line 4, in <module>
    from flask_security import uia_username_mapper
  File "/usr/local/lib/python3.10/dist-packages/flask_security/__init__.py", line 14, in <module>
    from .changeable import admin_change_password
  File "/usr/local/lib/python3.10/dist-packages/flask_security/changeable.py", line 19, in <module>
    from .utils import config_value, hash_password, login_user, send_mail
  File "/usr/local/lib/python3.10/dist-packages/flask_security/utils.py", line 42, in <module>
    from flask_wtf import csrf
  File "/usr/lib/python3/dist-packages/flask_wtf/__init__.py", line 3, in <module>
    from .csrf import CSRFProtect, CsrfProtect
  File "/usr/lib/python3/dist-packages/flask_wtf/csrf.py", line 10, in <module>
    from werkzeug.security import safe_str_cmp
ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security' (/usr/local/lib/python3.10/dist-packages/werkzeug/security.py)
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***
uWSGI running as root, you can use --uid/--gid/--chroot options
[... 1 lines cut ...]
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 9247)
spawned uWSGI worker 1 (pid: 9481, cores: 2)
spawned uWSGI worker 2 (pid: 9482, cores: 2)
spawned uWSGI worker 3 (pid: 9484, cores: 2)
spawned uWSGI worker 4 (pid: 9486, cores: 2)
spawned uWSGI worker 5 (pid: 9487, cores: 2)
*** Stats server enabled on 127.0.0.1:9191 fd: 17 ***
[2022-09-26 10:39:07][compare][INFO]: Plug-ins available: dict_keys(['File_Coverage', 'File_Header', 'Software'])
[... cut ...]
[2022-09-26 10:40:07][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0

When I access 127.0.0.1:5000, I get this. I am guessing it is one of the side effect of Werkzeug error.

--- no python application found, check your startup logs for errors ---
[pid: 9481|app: -1|req: -1/1] 127.0.0.1 () {40 vars in 644 bytes} [Mon Sep 26 10:40:25 2022] GET / => generated 21 bytes in 0 msecs (HTTP/1.1 500) 2 headers in 83 bytes (0 switches on core 0)

These are the current versions of Flask and Werkzeug
[image]

jstucke, 2022-09-28:
Some of the pip packages actually start with a capital "F". Therefore, the output of `pip list | grep -i flask` would be interesting.

> File "/usr/local/lib/python3.10/dist-packages/flask_security/utils.py", line 42, in
from flask_wtf import csrf
File "/usr/lib/python3/dist-packages/flask_wtf/init.py", line 3, in
from .csrf import CSRFProtect, CsrfProtect
File "/usr/lib/python3/dist-packages/flask_wtf/csrf.py", line 10, in
from werkzeug.security import safe_str_cmp

This still sounds like an incompatibility between some flask packages. 
For the record: I have the following versions installed:

```
> pip freeze | grep -iE "werkzeug|flask"
Flask==2.1.2
Flask-Login==0.6.1
flask-paginate==2022.1.8
Flask-Principal==0.4.0
flask-restx==0.5.1
Flask-Security-Too==4.1.4
Flask-SQLAlchemy==2.5.1
Flask-WTF==1.0.0
Werkzeug==2.1.2
```

Do you use a virtualenv? It's probably a good idea to use one to avoid dependency (version) conflicts but if you want to switch to a virtualenv you would need to run the installation again.

jstucke, 2022-10-17:
Any update on this? Did you get it running?

jstucke, 2022-10-20:
I will close this for now. If this is still an issue, please feel free to reopen it.

closed 2022-10-20 by jstucke