fkie-cad/FACT_core #859: faced some problem with database

Oddluo, 2022-09-22:
hey,I encountered this error after closing FACT and restarting, it seems like the problem with the database, Have you ever met？What should i do ?  Do I need to reinstall？
[image]

dorpvom, 2022-09-23:
Hi Oddlou,
this is quite a unique error and I've only seen similar errors when the RAM runs full (and objects might not be stored correctly anymore). I would cook up a patch to add some debug information if you can reproduce the error. Have you tried reproducing it yet?

Oddluo, 2022-09-26:
> 

hello，I fixed it by restarting it several times, and i don't know why ,maybe just a bug？ I can't reproduce it and i will closed this issue , thx.

closed 2022-09-26 by Oddluo