fkie-cad/FACT_core #860: ERROR: Could not connect to clamd on LocalSocket /var/run/clamav/clamd.ctl: No such file or directory

emorasoul, 2022-09-23:
when i analyzing a firmware，a error report frequently！I don't know what happy？
[image]

dorpvom, 2022-09-23:
Hi emorasoul,
I can't say why but it's coming from the malware detection plugin. As the plugin is marked for deprecation, we'll probably don't invest a lot into investigation but we'll look if we can make sense of the message.

jstucke, 2022-09-28:
It seems other people had similar problems with ClamAV and were able to resolve them: https://askubuntu.com/questions/1170774/clamav-clamd-ctl-file-is-not-getting-created-on-ubuntu

jstucke, 2022-10-17:
Incidentally "resolved" by removal of the malware scanner plugin #732

closed 2022-10-17 by jstucke